genodelabs/genode #4741: grub2: add default font

alex-ab, 2023-01-25:
To avoid scrambled output when booted in graphical mode and something went wrong (due to corrupted usb stick, not found partitions, invalid grub.cfg and so on), a font is required to see diagnostics and or the menu.

alex-ab, 2023-01-26:
The commit now adds to the contrib/g2fg sources a small font (from ttf-bitstream-vera-1.10 as already used by Genode), which is about 7kB unpacked and is sufficient for emergency cases. Since the font is just a file in the partition, it may be customized/replaced by anyone with something elase, if the need persists.

chelmuth, 2023-01-26:
I see the commit touch *tool/run/boot_dir/nova* but no other kernel. Shouldn't other boot scripts also be adapted?

alex-ab, 2023-01-27:
> I see the commit touch _tool/run/boot_dir/nova_ but no other kernel. Shouldn't other boot scripts also be adapted?

Yes. I added further commits which factor out the common parts, similar as already done for the iso case. Please check.

chelmuth, 2023-02-27:
Merged to master.

closed 2023-02-27 by chelmuth