genodelabs/genode #4746: tool/run: include /usr/sbin in search path

nfeske, 2023-01-27:
On Debian, I encountered the problem that some tools - such as resize2fs - reside in /usr/sbin, which is not included in the PATH of an ordinary user. So run-tool plugins like image/disk fail unless the user manually extends the PATH.

How about appending /usr/sbin to PATH environment variable in the run tool by default?

chelmuth, 2023-01-30:
The actual location of tools is quite distro-specific, for example, on Debian GNU/Linux 9.13 (stretch) *resize2fs* is installed in */sbin*. So, */sbin* should be added to `PATH` too.

cproc, 2023-01-30:
Would `exec [installed_command resize2fs]` instead of `exec resize2fs` solve the problem too?

nfeske, 2023-02-02:
That is the way! Thanks @cproc.

chelmuth, 2023-02-27:
Merged to master.

closed 2023-02-27 by chelmuth