genodelabs/genode #4755: base-hw: "Assertion failed: _head"

nfeske, 2023-02-08:
While stress-testing Sculpt OS on the PinePhone, I observed the following assertion.

```
Error: Assertion failed: _head
Error:   File: /plain/no/genode.git/depot/nfeske_test/src/base-hw-pinephone/2023-01-30/src/core/kernel/cpu_scheduler.cc:318
Error:   Function: Kernel::Cpu_share& Kernel::Cpu_scheduler::head() const 
```

nfeske, 2023-02-08:
Thanks to @skalk for commit https://github.com/nfeske/genode/commit/cc5f05d3299bac1405b3b6e3b4feb08f2deaaf27 and @m-stein for commit https://github.com/nfeske/genode/commit/6dbbc769703aa1cd37e65ba8644bfc65e20c079b, which apparently solve this issue.

@m-stein as of now, both commits reside on my pinephone branch. I would very much like to see the problem fixed on staging. Could you advise how to proceed, e.g., pointing me to a staging-ready commit to cherry-pick?

m-stein, 2023-02-08:
@nfeske I just re-checked my assumption and still would say that it is very unlikely that https://github.com/nfeske/genode/commit/cc5f05d3299bac1405b3b6e3b4feb08f2deaaf27 has any effect because the return values of ``_claim_for_head`` and ``_fill_for_head`` pretty directly relate to the validity of ``_head``. I'd therefore vote for merging https://github.com/nfeske/genode/commit/6dbbc769703aa1cd37e65ba8644bfc65e20c079b until I get to rework the scheduler internals (currently scheduled for march in my personal road map).

m-stein, 2023-02-08:
However, if you hesitate to go without https://github.com/nfeske/genode/commit/cc5f05d3299bac1405b3b6e3b4feb08f2deaaf27, I can plan for further stress testing https://github.com/nfeske/genode/commit/6dbbc769703aa1cd37e65ba8644bfc65e20c079b stand-alone.

nfeske, 2023-02-08:
It's alright. I merged the commit as 21fafb911f8222686ec5301ff8b9d577e13a8248 with a slightly changed commit title to make it fit 50 characters. @m-stein do you approve?

m-stein, 2023-02-08:
@nfeske Yep. Thanks ;)

chelmuth, 2023-02-27:
Merged to master.

closed 2023-02-27 by chelmuth