genodelabs/genode #4756: Broken USB device name

rolandbaer, 2023-02-08:
In the usb_block_drv the log message "Found USB device:" shows a broken manufacturer and product name like "ds2o (ds2oCASCI)" instead of "disk2go (disk2go CLASSIC II)".

It looks like the encoding is wrong.

chelmuth, 2023-02-09:
As suspected in #4757, the issue is caused by the use of the `usb_string()` Linux kernel function in *dde_linux/src/lib/lx_emul/usb.c*. From my perspective it would be better to fix it there and return an unmodified string descriptor to a STRING request (maybe retrieved by `usb_get_descriptor()`).

@ssumpf what's your opinion?

ssumpf, 2023-02-09:
@chelmuth: We wrote the `usb_block_drv` ourselves, it's not dde_linux. All strings given by USB hardware should be UTF-16 (as defined by the standard). Most of the times it should be sufficient to output the lower byte as UTF-8 and skip the upper one. One how this is implemented in `usb_block_drv` we would have to ask @cnuke.

chelmuth, 2023-02-09:
@ssumpf Don't you agree *repos/dde_linux/src/lib/lx_emul/usb.c* used by *pc_usb_host_drv* should return a string descriptor with UTF-16 strings on a `Usb::Packet_descriptor::STRING` request (see *repos/os/include/usb_session/usb_session.h*)?

cnuke, 2023-02-09:
The `usb_block_drv` driver uses the implementation from `Usb::String::to_char` from *usb/types.h*. Since this commit alteres the `string` type to *utf-8* that appears to solve the problem.

ssumpf, 2023-02-09:
@chelmuth: Okay, the `usb_string` function converts the descriptor to UTF-8 (https://elixir.bootlin.com/linux/v5.14.21/source/drivers/usb/core/message.c#L967) `Usb::String` more or less a nop. One could use the `usb_get_descriptor` function in order to retrieve a raw string descriptor (even though the comment says otherwise). Is there a reason we require UTF-16 for strings? Things like VirtualBox6 well get raw descriptors through control messages, and therefore, receive UTF-16 strings. Who else needs it?

chelmuth, 2023-02-09:
The answer to your question is a matter of design - the design of the USB session interface. In my opinion, the USB session should not provide any convenience regarding information specified in the USB spec. Therefore, it is questionable that it provides the (vaguely-specifed) STRING request at all. If such requests must be supported for some reason, then it must not convert any data but provide the raw USB request results. In case of string descriptors this means UTF-16.

rolandbaer, 2023-02-10:
From this discussion I see three solutions suggested:
- take the fix as it fixes the problem
- rewrite the fix to use UTF-16 strings
- throw out the strings at all

For me each of the first two options are OK and I can try to provide a PR for the second options and also test it with my raspi4 test setup. To drop the strings would not be my favorite option as I find it helpful to see a name and no more need to know vendor and product id's by heart.

So how do we go on from here?

chelmuth, 2023-02-10:
As first step I suggest to keep the special STRING request but use UTF-16 as in the USB standard (resp. remove the implicit conversion). The rationale is: The USB session provides a low-level access to USB devices. This requires changing *dde_linux/src/lib/lx_emul/usb.c* (maybe by patching linux to provide `usb_string_sub()` in *message.c* externally) and keep the client side as is.

In a second step, we should remove the string convenience from the host driver as it blocks for the completion of a control message. This could easily be implemented on client side (as it's done in libusb) as a utility.

rolandbaer, 2023-02-24:
I've created a second pull request that handles the string as utf-16.

The downside of the second version is that currently only UTF-16 up to 0xff are handled correctly (see sample at the end of this comment). This could be fixed by changing  the Usb::String implementation in genode/repos/os/include/usb/types.h. But this is only a academic issue, as I was not able to find a real device that used manufacture or product strings with characters outside the ascii range.

Samples:
PR 4757: Found USB device: Roland Bär (PI4→USB Deviçe) block size: 512 count: 2097152
PR 4772: Found USB device: Roland Bär (PI4USB Deviçe) block size: 512 count: 2097152

commit 661bbe69 (2023-03-03): usb_host: deliver UTF-16 strings on request

Linux kernel static functions usb_string_sub() and usb_get_langid() were
made accessible to implement robust string rerieval.

Fixes #4756
Fixes #4757
Fixes #4772
commit b9b18c92 (2023-03-13): usb_host: deliver UTF-16 strings on request

Linux kernel static functions usb_string_sub() and usb_get_langid() were
made accessible to implement robust string rerieval.

Fixes #4756
Fixes #4757
Fixes #4772
closed 2023-03-13 by chelmuth