genodelabs/genode #4759: hw: prepare ARMv8 EL2 mode appropriatedly to run Rockchip 3588

skalk, 2023-02-10:
@mickenx has enabled a first version of Genode's hw kernel running on RK3588. Therefore, it was necessary to zero out HCR and CPTR of EL2. He has done this within the initial assembler code that is run more or less unconditionally. In the generic codebase we cannot do so, because we do not necessarily start in EL2 mode. I wonder whether it is enough to zero these two registers within bootstrap's high-level code in Platform::enable_mmu() at the right place.

skalk, 2023-02-10:
@mickenx I've prepared commit 9c6b97e that shall also solve your initial bootstrap problems without replacing our crt0.so. I would be happy, if you could try it on your hardware target. In case it works, I would push it upstream. I did not zeroed the general-purpose and FPU registers. AFAIK there is no need in calling conventions to hand over zeroed registers, did you really need that to proceed?

skalk, 2023-02-10:
@mickenx I had to re-merge the commit to Genode's master branch, the staging one had problems, the new commit id is 0e6f390.

mickenx, 2023-02-10:
I will not be able to test this right away because of health issues. But will do in a week or so.

skalk, 2023-02-10:
@mickenx please take the time you need. There is no need to hurry up. Get well soon!

mickenx, 2023-02-23:
It works! Thanks.

skalk, 2023-02-23:
@mickenx thanks for testing.

chelmuth, 2023-02-27:
Merged to master.

closed 2023-02-27 by chelmuth