genodelabs/genode #4761: platform_drv: add support for dma_guard devices

jschlatow, 2023-02-10:
For Zynq, I have developed a dma_guard IP core which acts as an MPU for DMA transactions initiated by the FPGA. This design nicely complements the Platform interface, as the latter already makes the allocation of DMA buffers explicit. In consequence, the platform driver is able to take control over dma_guard devices and configure the DMA ranges accordingly whenever DMA buffers are allocated/freed.

On the downside, there are several mechanisms missing and challenges to be solved in the generic platform driver in order to achieve this:
- The platform driver must be able to acquire certain devices specified in the devices ROM.
- Since a dma_guard device is not global but corresponds to a single or a subset of  devices, there must be a way to specify the relation in the devices ROM.
- Whenever a device that requires the configuration of a dma_guard device is acquired/released, we might need to add/remove the already allocated DMA buffer ranges from the dma_guard.
- Moreover, when the devices ROM or policy ROM changes, dma_guard devices might need to be reconfigured.

jschlatow, 2023-02-10:
@skalk, I pushed a commit series for your review. Please take your time. Let me know if you need further explanations.

* b1aa8a5 fixes an issue with handling policy ROM updates
* 05bf581 generalises the concept of device ownership so that the platform driver itself can act as a device owner
* 66b6b5b introduces the concept of control devices, i.e. devices that are acquire by the platform driver itself
* 9d952e7 adds support for <control_device> node to the devices ROM so that devices can reference corresponding control devices
* 1aa1b12 is an intermediate commit to simplify the review, please feel free to squash it; it merely propagates the list of control devices from common to the session components
* c20f8aa applies the necessary modifications for configuring DMA ranges to the session component

jschlatow, 2023-02-10:
I also pushed two commits to my genode-zynq repo to demonstrate the utilisation of these changes by the zynq platform driver.

* jschlatow/genode-zynq@cdffd51 removes the hard-coded MMIO address for the system level control registers by making it a control device
* jschlatow/genode-zynq@4852a32 implements the dma_guard control device

jschlatow, 2023-02-20:
@skalk fyi, I'm going to mirror the platform-driver changes to the Zynq repo in order to feature the dma guard in the upcoming release. This leaves time for reviewing and testing in a wider scope until 23.05.

skalk, 2023-02-22:
@jschlatow thank you for following our commonly sketched plan so quickly. I was in holidays last week, but I will review the generic platform driver changes quite soon, at latest shortly after the release.

jschlatow, 2023-05-05:
@skalk I force-pushed a new commit series to my issue4761_platform_drv branch:

30ccbf8 and 85a4ac0 you already reviewed. 008cbc6 is the corresponding fixup.

bddeeb7 adds management of (platform-specific) IOMMU devices to common
d131ded adds the corresponding node to the devices ROM
265f953  is an intermediate commit to simplify the review, please feel free to squash it; it merely propagates the IOMMU devices from common to the session components
09a7cbf modifies the session component to control the IOMMU devices
b76044d transforms the Device_pd into an IOMMU device
7705a48 fixes an issue in base-nova that I stumbled upon

jschlatow, 2023-05-05:
I am currently running Sculpt 23.04 with the new platform driver. A ready-to-use image "sculpt-pc-2023-05-05" is available in my depot.

jschlatow, 2023-05-05:
There is another commit on my branch for which I forgot referencing this issue:

786f4d337efa7cc859992ce56a7167313297eece increases the RAM quota for the platform_session

skalk, 2023-05-15:
@jschlatow thank you very much for your carefully written commits! The only minor issue I've noticed is that some code blocks exceed the envisioned maximum of 80 characters per line, where it could be avoided. Apart from that I'm fine with it.

alex-ab, 2023-05-15:
> [7705a48](https://github.com/genodelabs/genode/commit/7705a48e9d20e806f418f89e59456695dc8dbd9b) fixes an issue in base-nova that I stumbled upon

Good catch. The question is, why one would need a partial page to be mapped ? 

If I understand correctly, your change would imply to not map the partial page at all, which on caller side may lead to a fault later on. In my opinion, the caller should know, whether it needs the whole page or the whole page not at all, or ?

jschlatow, 2023-05-15:
@skalk Thanks for your review. I pushed a few fixups commits to address the style issues.

alex-ab, 2023-05-15:
> > If I understand correctly, your change would imply to not map the partial page at all, which on caller side may lead to a fault later on. In my opinion, the caller should know, whether it needs the whole page or the whole page not at all, or ?
> 
> No, the patch implies to map the entire page. The loop continues until at least the requested size has been mapped.

Thanks for clarification.

skalk, 2023-05-22:
@jschlatow thank you for the fixups!
@chelmuth @nfeske: I've squashed @jschlatow 's work and rebased it to current upstream staging. You can find it on my merge_to_staging branch. Please consider it for the upstream staging branch.

chelmuth, 2023-05-24:
@jschlatow `make -C build/arm_v7a BOARD=zynq_qemu drivers/platform` fails with the series. Did we miss required commits in genode-zynq?

jschlatow, 2023-05-24:
@chelmuth Sorry, I just pushed the relevant commits to the genode-zynq repo's staging branch.

nfeske, 2023-05-30:
Fixed in master.

closed 2023-05-30 by nfeske