genodelabs/genode #4762: dde_linux: work structs that depend on each other may deadlock

ssumpf, 2023-02-13:
In test cases we could trigger sporadic deadlocks when one work blocks, schedules another work to the same work queue, and waits for completion of the later work.

ssumpf, 2023-02-13:
Commit 723790f addresses the issue.

chelmuth, 2023-02-24:
@ssumpf in an offline discussion @cnuke expressed his opinion that the change in this commit may be a temporary fix for a symptom and the original cause is related to something called "recycle task/work". Could you please shed some light on this and (if appropriate) open a distinct issue to address the original issue?

ssumpf, 2023-02-24:
@chelmuth: The rescue (not recycle) task is programmed via a timeout when a work is scheduled. In case a work does not finish within the timeout, the rescue task will execute the next work from the queue. The rescue task might, therefore, fix this issue as well. But because timing is not reliable right now, I cannot decide this. It might still be a UP vs. MP problem. We will have to review this when timing works better. In the meantime the fix works well, because there will always be three tasks per work queue present, so in case two works depend on each other, the third can possibly resolve their suffering.

ssumpf, 2023-02-24:
@chelmuth: Btw, the commit message says more or less the same as the above.

commit c6c0c9d3 (2023-02-23): linux: allow more workers when scheduling works

Workers are only summoned by the manager or when there are no workers on
a work queue. In case a work in front of a work queue depends on a work
behind it, a deadlock may occur. To solve this Linux spawns a rescue
worker using timers. Timing of Linux based drivers is currently reworked
and not reliable. Therefore, we increase the number of workers that are
spawned from 1 to 3 per work queue in order to resolve possible
deadlocks.

Fixes #4762
closed 2023-02-27 by chelmuth