genodelabs/genode #4771: usb_block: enable effective C++ error switch

skalk, 2023-02-23:


skalk, 2023-02-23:
@cnuke please review commit 92c3991, thanks!

skalk, 2023-02-24:
@cnuke I've updated the commit to ee7c86e to add a comment to the maximum block count (CMD 10 vs. 16) condition, and to make that condition more clear to newcomers.

cnuke, 2023-02-24:
@skalk thanks your doing the clean up and in doing so catching the block count bug. I am fine with your changes.

(FWIW, I noticed that the `usb_block.run` run-script does not work anymore as the *default-policy* is not enough to access the device due to the required devce label in case of using a *class-wildcard* policy. Commit ec9cd04 illustrates a workaround but I am not sure if that is a feature we actually want to support or if the run-script should be made more specific, e.g. only executable on Qemu with the USB stick always attached to `usb-1-2`.)