genodelabs/genode #4774: dde_linux: separate virt_linux from wireguard

chelmuth, 2023-02-24:
In preparation for further hardware-agnostic components, e.g., updated lxip or lx_emul tests, virt_linux must represent a real platform not interwoven with the wireguard port.

chelmuth, 2023-02-24:
The commits above implement the idea. As *dde_linux/recipes/src/wireguard* was not tested yet it certainly misses required adaptation, which be added in a fixup commit.

@m-stein what's your opinion about the changes?

m-stein, 2023-03-03:
@chelmuth I'm sorry for the delayed response, your message slipped through on my side. I'm not too much into the details anymore but the commit looks fine to me. Thans for having taken the effort to make this clean separation! I have found only some minor issues:

* Is it necessary to postfix the architecture-dependent dummies files with the architecture name? It feels redundant to me.
* device.h and irq.cc already missed beforehand and still miss a comment header.
* Is it necessary to re-order items in MIRRORED_FROM_REP_DIR in content.mk and SRC_C in wireguard.inc? Without it, the commit would be a bit easier to read.

commit 3a9ef487 (2023-04-17): dde_linux: separate virt_linux from wireguard

Preparation to support further hardware-agnostic dde_linux components
(incl. lxip and tests).

Fixes #4774
closed 2023-04-17 by chelmuth