genodelabs/genode #4776: Browser-dependent performance variability in Linux VM on Sculpt

jschlatow, 2023-02-28:
I recently noticed that my private nextcloud installation performs very slowly when requested from a qtwebengine-based browser within a VM running on Sculpt. Especially the Files app is barely usable, e.g. when typing the name of a new folder in the web UI, the characters appear with a lag of approx. 1s. Interestingly, the web UI runs smoothly when opened in firefox in the same VM. Also, booting the same Linux system natively instead of in a VM on Sculpt, the qtwebengine-based browsers offer the same smooth browsing experience.

The described behaviour is reproducible with the offical demo at https://try.nextcloud.com/.

I'd like to investigate this issue. Does anyone have a guess where to look at?

chelmuth, 2023-03-01:
Could it be your scenario experiences excessive VM exits?

Do you use a patch like 5dee7010a322516cf922764084c021ad270ff6a6 which trades entirely virtualized cycle counting against elimination of RDTSC exit overhead?

For analysis of vbox6 exits you may give c3c4303606067d324c696794b6e68d14aa47656b and 8b914efdc143eac7efd1563ba1694e3f965a9b77 a try.

cproc, 2023-03-01:
I can reproduce this symptom with Falkon in a Xubuntu 22.04 VM when the window of the VM is maximized on a 1920x1080 display. It's even worse with the Chromium browser where it already takes many seconds to enter a URL. I can confirm that Firefox doesn't seem to have this symptom. But when the VM window has its original size of 1024x768 the symptom seems to disappear for me.

jschlatow, 2023-03-06:
I can confirm that this seems to be somehow related to the window size. I experimented with different sizes. At first, it seemed as if the performance degrades with increasing area. However, some resolutions perform significantly worse than others, e.g. at 1680x1050 there is a noticable delay when typing a new folder name whereas, at 1600x1050 and 1700x1050, the delay disappears or at least is less noticeable.

jschlatow, 2023-03-06:
@chelmuth I cherry-picked 5dee7010a and c3c4303. Can you help me interpret whether the logged counts correspond to a normal or excessive number of VM exits? The log below shows a few seconds before and after maximizing the VM window. In both cases, I have been typing a few letters in the nextcloud UI.

```
[runtime -> archlinux_dbg -> vbox] [1] total=400000 virt_exit {
[runtime -> archlinux_dbg -> vbox] [1]  49207 7        VMX_EXIT_INT_WINDOW - 7 - Interrupt window.
[runtime -> archlinux_dbg -> vbox] [1]  5112 10        VMX_EXIT_CPUID - 10 - CPUID instruction.
[runtime -> archlinux_dbg -> vbox] [1]  33189 12        VMX_EXIT_HLT - 12 - HLT instruction.
[runtime -> archlinux_dbg -> vbox] [1]  4 18        VMX_EXIT_VMCALL - 18 - VMCALL instruction.
[runtime -> archlinux_dbg -> vbox] [1]  6 29        VMX_EXIT_MOV_DRX - 29 - Debug-register accesses.
[runtime -> archlinux_dbg -> vbox] [1]  22845 30        VMX_EXIT_PORT_IO - 30 - I/O instruction.
[runtime -> archlinux_dbg -> vbox] [1]  20 31        VMX_EXIT_RDMSR - 31 - RDMSR instruction.
[runtime -> archlinux_dbg -> vbox] [1]  16 32        VMX_EXIT_WRMSR - 32 - WRMSR instruction.
[runtime -> archlinux_dbg -> vbox] [1]  245629 48        VMX_EXIT_EPT_VIOLATION - 48 - EPT violation.
[runtime -> archlinux_dbg -> vbox] [1]  1 54        VMX_EXIT_WBINVD - 54 - WBINVD instruction.
[runtime -> archlinux_dbg -> vbox] [1]  1 254        <null-string>
[runtime -> archlinux_dbg -> vbox] [1]  43970 255        <null-string>
[runtime -> archlinux_dbg -> vbox] [1] }
[runtime -> archlinux_dbg -> vbox] [1] rdmsr {
[runtime -> archlinux_dbg -> vbox] [1]  0xc0011029 : 1
[runtime -> archlinux_dbg -> vbox] [1]  0x64e : 1
[runtime -> archlinux_dbg -> vbox] [1]  0xce : 1
[runtime -> archlinux_dbg -> vbox] [1]  0x8b : 1
[runtime -> archlinux_dbg -> vbox] [1]  0x17a : 1
[runtime -> archlinux_dbg -> vbox] [1]  0x1b : 1
[runtime -> archlinux_dbg -> vbox] [1]  0x3a : 1
[runtime -> archlinux_dbg -> vbox] [1]  0x10a : 1
[runtime -> archlinux_dbg -> vbox] [1]  0x140 : 1
[runtime -> archlinux_dbg -> vbox] [1]  0xc0000080 : 2
[runtime -> archlinux_dbg -> vbox] [1]  0x34 : 1
[runtime -> archlinux_dbg -> vbox] [1]  0x179 : 4
[runtime -> archlinux_dbg -> vbox] [1]  0x1a0 : 4
[runtime -> archlinux_dbg -> vbox] [1] }
[runtime -> archlinux_dbg -> vbox] [1] wrmsr {
[runtime -> archlinux_dbg -> vbox] [1]  0x140 : 1
[runtime -> archlinux_dbg -> vbox] [1]  0x3a : 1
[runtime -> archlinux_dbg -> vbox] [1]  0xc0000080 : 1
[runtime -> archlinux_dbg -> vbox] [1]  0x174 : 1
[runtime -> archlinux_dbg -> vbox] [1]  0xc0000102 : 1
[runtime -> archlinux_dbg -> vbox] [1]  0xc0000101 : 2
[runtime -> archlinux_dbg -> vbox] [1]  0xc0000103 : 1
[runtime -> archlinux_dbg -> vbox] [1]  0xc0000081 : 1
[runtime -> archlinux_dbg -> vbox] [1]  0x175 : 1
[runtime -> archlinux_dbg -> vbox] [1]  0x8b : 1
[runtime -> archlinux_dbg -> vbox] [1]  0xc0000082 : 1
[runtime -> archlinux_dbg -> vbox] [1]  0x176 : 1
[runtime -> archlinux_dbg -> vbox] [1]  0xc0000084 : 1
[runtime -> archlinux_dbg -> vbox] [1]  0xc0000100 : 1
[runtime -> archlinux_dbg -> vbox] [1]  0x4b564d01 : 1
[runtime -> archlinux_dbg -> vbox] [1] }
[runtime -> archlinux_dbg -> vbox] [0] total=1400000 virt_exit {
[runtime -> archlinux_dbg -> vbox] [0]  21476 7        VMX_EXIT_INT_WINDOW - 7 - Interrupt window.
[runtime -> archlinux_dbg -> vbox] [0]  863630 10        VMX_EXIT_CPUID - 10 - CPUID instruction.
[runtime -> archlinux_dbg -> vbox] [0]  14724 12        VMX_EXIT_HLT - 12 - HLT instruction.
[runtime -> archlinux_dbg -> vbox] [0]  6 18        VMX_EXIT_VMCALL - 18 - VMCALL instruction.
[runtime -> archlinux_dbg -> vbox] [0]  1 28        VMX_EXIT_MOV_CRX - 28 - Control-register accesses.
[runtime -> archlinux_dbg -> vbox] [0]  6 29        VMX_EXIT_MOV_DRX - 29 - Debug-register accesses.
[runtime -> archlinux_dbg -> vbox] [0]  286410 30        VMX_EXIT_PORT_IO - 30 - I/O instruction.
[runtime -> archlinux_dbg -> vbox] [0]  40 31        VMX_EXIT_RDMSR - 31 - RDMSR instruction.
[runtime -> archlinux_dbg -> vbox] [0]  22 32        VMX_EXIT_WRMSR - 32 - WRMSR instruction.
[runtime -> archlinux_dbg -> vbox] [0]  194720 48        VMX_EXIT_EPT_VIOLATION - 48 - EPT violation.
[runtime -> archlinux_dbg -> vbox] [0]  1 254        <null-string>
[runtime -> archlinux_dbg -> vbox] [0]  18964 255        <null-string>
[runtime -> archlinux_dbg -> vbox] [0] }
[runtime -> archlinux_dbg -> vbox] [0] rdmsr {
[runtime -> archlinux_dbg -> vbox] [0]  0x64d : 1
[runtime -> archlinux_dbg -> vbox] [0]  0x641 : 1
[runtime -> archlinux_dbg -> vbox] [0]  0x611 : 1
[runtime -> archlinux_dbg -> vbox] [0]  0xfe : 3
[runtime -> archlinux_dbg -> vbox] [0]  0x277 : 1
[runtime -> archlinux_dbg -> vbox] [0]  0x1a0 : 6
[... 152 lines cut ...]
[runtime -> archlinux_dbg -> vbox] [1]  0x140 : 1
[runtime -> archlinux_dbg -> vbox] [1]  0xc0000080 : 2
[runtime -> archlinux_dbg -> vbox] [1]  0x34 : 1
[runtime -> archlinux_dbg -> vbox] [1]  0x179 : 4
[runtime -> archlinux_dbg -> vbox] [1]  0x1a0 : 4
[runtime -> archlinux_dbg -> vbox] [1] }
[runtime -> archlinux_dbg -> vbox] [1] wrmsr {
[runtime -> archlinux_dbg -> vbox] [1]  0x140 : 1
[runtime -> archlinux_dbg -> vbox] [1]  0x3a : 1
[runtime -> archlinux_dbg -> vbox] [1]  0xc0000080 : 1
[runtime -> archlinux_dbg -> vbox] [1]  0x174 : 1
[runtime -> archlinux_dbg -> vbox] [1]  0xc0000102 : 1
[runtime -> archlinux_dbg -> vbox] [1]  0xc0000101 : 2
[runtime -> archlinux_dbg -> vbox] [1]  0xc0000103 : 1
[runtime -> archlinux_dbg -> vbox] [1]  0xc0000081 : 1
[runtime -> archlinux_dbg -> vbox] [1]  0x175 : 1
[runtime -> archlinux_dbg -> vbox] [1]  0x8b : 1
[runtime -> archlinux_dbg -> vbox] [1]  0xc0000082 : 1
[runtime -> archlinux_dbg -> vbox] [1]  0x176 : 1
[runtime -> archlinux_dbg -> vbox] [1]  0xc0000084 : 1
[runtime -> archlinux_dbg -> vbox] [1]  0xc0000100 : 1
[runtime -> archlinux_dbg -> vbox] [1]  0x4b564d01 : 1
[runtime -> archlinux_dbg -> vbox] [1] }

```

chelmuth, 2023-03-06:
The number in `[1]` is the CPU ID. As the other CPUs are not part of the log let's investigate CPU [1] that had 200.000 exits during the test period. Depending on the real time elapsed this could be interpreted as *a lot of exits*. From the 200K, 107862 are EPT violations (maybe accesses to unmapped video memory) plus 25K HLT exits (CPU went idle). There are also 20K interrupt windows + 13K pause exits for IRQ injection.

The question is now: How does it look in a similar time frame without performance degradation?

jschlatow, 2023-03-06:
I re-did the measurements, once with a maximized window (performance degradation) and once at the default size (no performance degradation). In both cases, it took approx. 50-60s for 100k exits on CPU 1. The invididual numbers are also pretty similar, yet there are about one third more interrupt windows.

|  | fast | slow |
|--|------|--------|
| INT_WINDOW | 8980 | 12127 |
| HLT | 13149 | 11118 |
| PORT_IO | 15268 | 15070 |
| EPT_VIOLATION | 56626 | 53524 |

jschlatow, 2023-03-06:
The numbers are also similar when using firefox (maximized VM window):

| | firefox | qutebrowser |
|-|----|----|
| INT_WINDOW | 10083 | 8980 |
| HLT | 10942 | 13149 |
| PORT_IO | 13805 | 15268 |
| EPT_VIOLATION | 57601 | 56626 |

Based on these numbers, I think we can dismiss the VM-exit theory.

chelmuth, 2023-03-07:
> Based on these numbers, I think we can dismiss the VM-exit theory.

Agreed.

cproc, 2023-03-08:
> @cproc Do you know of any switches for QtWebEngine resp. Falkon to disable GPU acceleration?

I know these options: `QT_QUICK_BACKEND=software`, `--disable-gpu` and `--disable-gpu-compositing`.

I just tried these on my x260 with the Xubuntu 22.04 VM and Falkon, but it did not help here and I also still got the lag with Chromium (same version 110.0.5481.177).

closed 2023-04-11 by jschlatow