genodelabs/genode #4778: lx_emul: adjust and unify time handling

skalk, 2023-03-03:
Time handling still isn't always done appropriatedly in the Linux device driver environment. The jiffies value sometimes is outdated too long. Every driver implements its own fashion of udelay, which isn't semantically correct  either.

ssumpf, 2023-03-04:
@skalk: This looks really promising. Good job!

chelmuth, 2023-03-06:
Current build/link errors in *app/wireguard* stem from commit 65e1948b98f816ddb16ea434e7e758695264cf8c and the following code in contrib linux.

```c
static __must_check inline bool may_use_simd(void)
{
	return !in_interrupt();
}
```

skalk, 2023-03-06:
Okay, that means the former semantic of `in_interrupt` returned true, although this was wrong. Now, it tries to use optimized SIMD instructions. We should simply add them.

chelmuth, 2023-03-13:
Merged to master.

closed 2023-03-13 by chelmuth