genodelabs/genode #4779: platform_drv: dies with unhandled exception in Sculpt

alex-ab, 2023-03-04:
Using Sculpt on Qemu triggered several times the following exception (1 out of 4 starts): 
```
Error: I/O memory [00000000b0000000,00000000b0001000) not available
Error: Local MMIO mapping failed!
[init -> drivers -> platform_drv] Error: IO_MEM-session creation failed (cap_quota=3, ram_quota=6144, base=0xb0000000, size=0x1000, wc=no)
[init -> drivers -> platform_drv] Error: Uncaught exception of type 'Genode::Service_denied'
```

alex-ab, 2023-03-04:
According to the log, the pci_decode in parallel to the platform driver uses the I/O memory. It seems that pci decoder parses the PCI devices more than once ...

```
...
[init -> drivers -> acpi_drv] Found MADT
[init -> drivers -> acpi_drv] MADT IRQ 0 -> GSI 2 flags: 0
[init -> drivers -> acpi_drv] MADT IRQ 5 -> GSI 5 flags: 13
[init -> drivers -> acpi_drv] MADT IRQ 9 -> GSI 9 flags: 13
[init -> drivers -> acpi_drv] MADT IRQ 10 -> GSI 10 flags: 13
[init -> drivers -> acpi_drv] MADT IRQ 11 -> GSI 11 flags: 13
[init -> drivers -> usb_hid_drv] --- USB HID input driver ---
[init -> drivers -> usb_hid_drv] Warning: use compatibility mode: will claim all HID devices from USB report
[init -> drivers -> usb_hid_drv] Configured HID screen with 0x0 (multitouch=false)
[init -> drivers -> acpi_drv] Found MCFG
[init -> drivers -> acpi_drv] MCFG BASE 0xb0000000 seg 0x0 bus 0x0-0xff
[init -> drivers -> acpi_drv] RSDT OEM 'BOCHS ', table id 'BXPCRSDT', revision 1, creator 'BXPC' (1)
[init -> drivers -> acpi_drv] SMBIOS table (entry point: 0x1658d0 structures: 0xf58f0)
[init -> drivers -> pci_decode] Error: parse pci start ...
Error: _prepare_io_mem 0x6c2018 base=0xb0000000, size=0x100000, wc=no, diag=0, label="init -> drivers -> pci_decode -> ", ram_quota=3726, cap_quota=2
Error: ~Io_mem_session_component 0x6c2018
[init -> runtime] parent provides
[init -> runtime]   service "VM"
[init -> runtime]   service "Usb"
[init -> runtime]   service "File_system"
[init -> runtime]   service "Rtc"
[init -> runtime]   service "TRACE"
[init -> runtime]   service "Event"
[init -> runtime]   service "Capture"
[init -> runtime]   service "Gpu"
[init -> runtime]   service "Pin_state"
[init -> runtime]   service "Pin_control"
[init -> runtime] child "runtime_view"
[init -> runtime]   RAM quota:  7944K
[init -> runtime]   cap quota:  166
[init -> runtime]   ELF binary: menu_view
[init -> runtime]   priority:   0
[init -> runtime] child "panel_view"
[init -> runtime]   RAM quota:  3848K
[init -> runtime]   cap quota:  116
[init -> runtime]   ELF binary: menu_view
[init -> runtime]   priority:   0
[init -> runtime] child "menu_view"
[init -> runtime]   RAM quota:  3848K
[init -> runtime]   cap quota:  116
[init -> runtime]   ELF binary: menu_view
[init -> runtime]   priority:   0
[init -> runtime] child "settings_view"
[init -> runtime]   RAM quota:  3848K
[init -> runtime]   cap quota:  116
[init -> runtime]   ELF binary: menu_view
[init -> runtime]   priority:   0
[init -> runtime] child "network_view"
[init -> runtime]   RAM quota:  3848K
[init -> runtime]   cap quota:  116
[init -> runtime]   ELF binary: menu_view
[init -> runtime]   priority:   0
[init -> runtime] child "popup_view"
[init -> runtime]   RAM quota:  3848K
[init -> runtime]   cap quota:  116
[init -> runtime]   ELF binary: menu_view
[init -> runtime]   priority:   0
[init -> runtime] child "file_browser_view"
[init -> runtime]   RAM quota:  7944K
[init -> runtime]   cap quota:  116
[init -> runtime]   ELF binary: menu_view
[init -> runtime]   priority:   0
[init -> runtime] child "ram_fs"
[init -> runtime]   RAM quota:  776K
[init -> runtime]   cap quota:  266
[init -> runtime]   ELF binary: vfs
[init -> runtime]   priority:   0
[init -> runtime]   provides service File_system
[init -> runtime] child "ram_fs" announces service "File_system"
[init -> drivers -> pci_decode] Error: parse pci done
[init -> drivers -> pci_decode] Error: parse pci start ...
Error: _prepare_io_mem 0x6d1018 base=0xb0000000, size=0x100000, wc=no, diag=0, label="init -> drivers -> pci_decode -> ", ram_quota=3726, cap_quota=2
Error: I/O memory [00000000b0000000,00000000b0001000) not available
Error: Local MMIO mapping failed!
[init -> drivers -> platform_drv] Error: IO_MEM-session creation failed (cap_quota=3, ram_quota=6144, base=0xb0000000, size=0x1000, wc=no)
[init -> drivers -> platform_drv] Error: Uncaught exception of type 'Genode::Service_denied'
Error: ~Io_mem_session_component 0x6d1018
[init -> drivers -> platform_drv] Warning: abort called - thread: ep
[init -> drivers -> pci_decode] Error: parse pci done

```

chelmuth, 2023-03-08:
The solution in 9d5d39abe217832e5cc0e2f7af2cf646d4bac67a is quite unexpected and I have to admit it looks over-pragmatic to my taste. Yet I'd merge the commit if it's just a short-term intermediate step to the integration of *pci_decode* into *pc_platform_drv*.  What do you think of that?

skalk, 2023-03-08:
@chelmuth I've overworked the commit to be less pragmatic.

skalk, 2023-03-08:
@chelmuth according to a hint by @nfeske again I've re-worked my commit to additionally exclude the ACPI ROM from the main object. The for now final version is commit fed287c

closed 2023-03-13 by chelmuth