genodelabs/genode #4780: Remove util/avl_string.h

nfeske, 2023-03-06:
With the `Dictionary` introduced in #4610 in place, the old `Avl_string` utility should be retired.

nfeske, 2023-03-09:
The commits above eliminates the use of `Avl_string` from the code base except for one remaining little VFS plugin, namely `vfs_trace`.

I tried and failed two times attempting to wrap my head around the code at https://github.com/genodelabs/genode/blob/master/repos/gems/src/lib/vfs/trace/directory_tree.h on my mission to decouple it from `Avl_string`. Both times I got lost in a forest of trees.

@ssumpf, could you please do the update of the code to the use of the Dictionary utility?

ssumpf, 2023-03-09:
@nfeske: 84bb68f is my take on a library full of dictionaries.

nfeske, 2023-03-10:
Thanks @ssumpf for updating the code so quickly.

commit fc6a9a4c (2023-03-10): base: remove util/avl_string.h

The use case is covered by util/dictionary.h now.

Fixes #4780
closed 2023-03-13 by chelmuth