genodelabs/genode #4781: Use Io_progress_handler only if necessary, and an Io_signal_handler is involved

skalk, 2023-03-08:
Io_progress_handler can help to reduce the number of outgoing signals, e.g. if a bunch of I/O operation is done by a driver due to an initial signal (irq, packet stream) but only at the end of all operations signals shall be send to its participants. However, this handler only gets called at the end of an Io_signal_handler routine. There are some places where an Io_progress_handler::handle_io_progress routine is empty. In those cases no Io_progress_handler shall be used at all. Other drivers copied the pattern, where an Io_progrss_handler is effectively used, but not all necessary signal sources are realized via Io_signal_handler.

skalk, 2023-03-08:
I've mixed up front- and backend signal sources, so only remove Io_progress_handler where it is obviously useless.

chelmuth, 2023-03-13:
Merged to master.

closed 2023-03-13 by chelmuth