genodelabs/genode #4785: vfs_pipe: large data may block after 23.02

trimpim, 2023-03-14:
After updating to 23.02 some of our tests failed on all platforms.

What is common in all the failed scenarios, is that about 1MiB of data is, that `stdin`, `stdout` and `stderr` of some ported `libc`  applications are redirected to a native component via a `vfs` server that has three separate `vfs_pipe` instances.

The problem occurs when the data on `stdout`/`stderr` exceeded ~64KiB. On `stdin`  the transfer of 1MiB went through without a problem.

When using `test-libc_fifo_pipe` to debug the issue, I discovered, that this failed when the buffer was increased. I created a separate commit, which fixes the test and increases its buffer size. Unfortunately the test did not show the same error.

Adding some debug messaged, I figured out, that the readers aren't notified when the writer pipe fills up. I have created a patch that fixes the issue for our tests. As I'm no expert in `VFS` I'm not sure if this is correct or if the fix should be done in an other way.

nfeske, 2023-03-14:
Thanks @trimpim for reporting and debugging this issue. Based on your description and the patch, I have come up with a small artificial test case (on branch https://github.com/nfeske/genode/commits/pipe_read_ready).

The problem appears when using select on the read-end of the pipe. Once the pipe becomes empty, the reader expects to be woken up from a select call. I wonder, does this presumption match your experience with the tests you mentioned?

In the scenario with the VFS server, the underlying select mechanics rely on the interplay of the vfs_pipe plugin, the vfs library, the vfs server, the file-system session, and the libc. While I removed the Io_response_handler from the VFS in version 23.02, I was a bit too eager in the vfs_pipe plugin.

The Read_ready_response_handler will eventually be removed from the VFS as a further simplification step (to be replaced with the much simpler Vfs::User hook). The test case will hopefully prevent another regression like this.

I merged both of your commits to staging.

trimpim, 2023-03-15:
@nfeske happy to help. Yes, the scenario you described is very close, what our application does.

I like the test you created. I assume, this will also go in to `master`. I would like to integrate this in to our testing pipeline.

I'm looking forward to the further simplification of VFS.

trimpim, 2023-03-15:
@nfeske sorry for missing this problem.

The fixup  8f1f94fb0f7a47ad4dd6d0fc7310e3a3d0984700 sets the buffer size to 16KiB. With this the the test passes for me on x86_32.

nfeske, 2023-03-15:
Thanks for the update. Both commits are back on staging now. ;-)

> I like the test you created. I assume, this will also go in to master. I would like to integrate this in to our testing pipeline.

It will probably be added to the depot-autopilot but this needs some further curation on my side. It turns out that the change of stdin2out (the commit preceding the actual test) is not a good idea because it breaks the use case of watching the log file in Sculpt. A file fd is always reported as immediately readable by select. So the sleepy polling loop cannot be replaced by the use of select.

nfeske, 2023-03-15:
Commit https://github.com/genodelabs/genode/commit/b888466d9c79fcc0a5fb283ebfe3b49b0b7ebb80 adds the test to the depot autopilot.

trimpim, 2023-03-16:
@nfeske I did cherry-pick your commit to our fork, to add it to the pipeline. When executing it on `arm_v8a`  the counter requested resources (CAPS).

with 6add7721bdfc2cc16af8cedacc08b7aba1bcc2fc the test also passes on that platform.

nfeske, 2023-03-16:
Thanks @trimpim for the fixup. :)

nfeske, 2023-04-19:
The fix entered the master branch.