genodelabs/genode #4786: part_block: split reporting from parsing

cnuke, 2023-03-14:


cnuke, 2023-03-14:
Commit [73c53c0](https://github.com/cnuke/genode/commit/73c53c0249773aa6ca5168d3cce879afa52b985a) seperates parsing from reporting.

nfeske, 2023-03-14:
Thanks a lot @cnuke for much appreciated rework.

As a minor further improvement, may you consider replacing the `generate_report` interface from

```
virtual void generate_report(Expanding_reporter &reporter)
```

to

```
virtual void generate_report(Xml_generator &xml) const
```
?

The `const` reinforces that the report-generation has indeed no side effects, and the use of the `Xml_generator` exposes only as much as actually needed.

cnuke, 2023-03-14:
No problem, the fixup commit [9603424](https://github.com/cnuke/genode/commit/960342498d53b555126683550ca2281bd13f861f) makes the *const* change.

nfeske, 2023-03-14:
Thanks.

Do you have reservations about the suggested change of the argument type?

cnuke, 2023-03-16:
@nfeske sorry, my bad - I overlooked changing the argument. I prepared commit  [7805c57](https://github.com/cnuke/genode/commit/7805c57b296b01f7d69ff486565fc6784f3a3f2c) that includes both suggestions.

cnuke, 2023-03-16:
Commit [8fcee8f](https://github.com/cnuke/genode/commit/8fcee8ffb40bf81e241f136d6f2b422be42d2eb4) contains fixups for the various remarkes.

commit f7b910b2 (2023-04-17): part_block: split reporting from parsing

Previously the 'partitions' report was implicitly generated after
parsing the partition-table by accessing the referenced reporter
object.

Now the report is explicitly created by calling the 'generate_report'
member function. At this point we no longer have access to all
required information so the 'Partition' class was extended to
accommodate for the data needed by MBR as well as GPT reports.

Fixes #4786.
closed 2023-04-17 by chelmuth