genodelabs/genode #4789: Missing 'pc_usb_host_drv' recipe

daihuasheng, 2023-03-16:
I'm trying to build genode for nova in Ubuntu:22.04 by executing: 
`make -C build/x86_64 KERNEL=nova BOARD=pc run/demo`
And if I uncomment the ` RUN_OPT += --depot-auto-update`, I will get an error: 
`Error: recipe for 'pc_usb_host_drv' is missing`
I noticed that issue #4593 had the same problem, but it was not solved eventually.
Furthermore, if I comment the `--depot-auto-update`, and try to execute 
`./tool/depot/create genodelabs/bin/x86_64/pc_usb_host_drv CROSS_DEV_PREFIX=/usr/local/genode/tool/21.05/bin/genode-x86`
I get this error in `Library pc_linux_generated`, during `CONFIG Linux`:
`[Linux]  /host/genode/depot/genodelabs/api/pc_linux/2023-03-12/src/linux/scripts/extract-cert.c:21:10: fatal error: openssl/bio.h: No such file or directory`
`[Linux]     21 | #include <openssl/bio.h>`
I have no idea whether these refer to the same problem, since I'm quite new for this. I'm sure that the genode code is the newest. Would you please supply any suggestions for handling this?

skalk, 2023-03-16:
Did you enabled resp. uncommented the necessary repositories in your `etc/build.conf` file?:
```
REPOSITORIES += $(GENODE_DIR)/repos/dde_linux
REPOSITORIES += $(GENODE_DIR)/repos/libports
REPOSITORIES += $(GENODE_DIR)/repos/pc
```

nfeske, 2023-03-16:
> fatal error: openssl/bio.h: No such file or directory

I guess that the Linux build system - that is invoked for preparing a Linux-kernel build directory - requires the libssl-dev package to be installed on your development machine.

daihuasheng, 2023-03-16:
> > fatal error: openssl/bio.h: No such file or directory
> 
> I guess that the Linux build system - that is invoked for preparing a Linux-kernel build directory - requires the libssl-dev package to be installed on your development machine.

You're right, I just presumed that "openssl" comes from some additional libraries. My fault.

daihuasheng, 2023-03-16:
> Did you enabled resp. uncommented the necessary repositories in your `etc/build.conf` file?:
> 
> ```
> REPOSITORIES += $(GENODE_DIR)/repos/dde_linux
> REPOSITORIES += $(GENODE_DIR)/repos/libports
> REPOSITORIES += $(GENODE_DIR)/repos/pc
> ```

Oh, I forgot to uncomment the "pc" repository. Now it works well. My apology for making such a stupid mistake. Thank you for the useful suggestions. Wish you two a great day!

closed 2023-03-16 by daihuasheng