genodelabs/genode #4792: Small allocations trigger anomaly in the x86 platform_drv

cnuke, 2023-03-22:
TL;DR calling `alloc_dma_buffer()` with a size of *1* leads to unexpected behaviour.

@chelmuth observed that executing the `nvme` driver in Qemu exhibits an unexpected delay:

```
12:25:27.660545 
                NOVA Microhypervisor v9-44e0d2d (x86_64): Mar 22 2023 12:24:05 [gcc 10.3.0] [MBI2]
12:25:27.660576 
12:25:27.676101 [ 0] TSC:2821292 kHz BUS:1004729 kHz (measured)
12:25:27.676666 [ 0] CORE:00:00:0 6:f:b:0 [0] Intel(R) Core(TM)2 Duo CPU     T7700  @ 2.40GHz
12:25:27.727490 
12:25:27.728466 Hypervisor NOVA (API v9)
[…]
12:25:27.785565 Genode 23.02-69-g9d17bb3cb9
12:25:27.793224 758 MiB RAM and 63253 caps assigned to init
12:25:28.484441 [init -> block_tester] --- start tests ---
12:25:28.526519 [init -> block_tester] start sequential READ start:0 size:65536 length:268435456 copy:1 batch:1
12:25:28.569616 [init -> acpi_drv] Found MADT
12:25:28.570045 [init -> acpi_drv] MADT IRQ 0 -> GSI 2 flags: 0
12:25:28.570399 [init -> acpi_drv] MADT IRQ 5 -> GSI 5 flags: 13
12:25:28.571482 [init -> acpi_drv] MADT IRQ 9 -> GSI 9 flags: 13
12:25:28.571502 [init -> acpi_drv] MADT IRQ 10 -> GSI 10 flags: 13
12:25:28.571508 [init -> acpi_drv] MADT IRQ 11 -> GSI 11 flags: 13
12:25:28.571913 [init -> acpi_drv] Found MCFG
12:25:28.572478 [init -> acpi_drv] MCFG BASE 0xb0000000 seg 0x0 bus 0x0-0xff
12:25:28.574111 [init -> acpi_drv] RSDT OEM 'BOCHS ', table id 'BXPCRSDT', revision 1, creator 'BXPC' (1)
12:25:28.675188 [init -> acpi_drv] SMBIOS table (entry point: 0x1558f0 structures: 0xf5910)
12:25:29.844636 [ 0] sys_assign_pci: Invalid Hint (0x18)
12:25:29.846861 [init -> platform_drv] Assignment of PCI device 00:03.0 to device PD failed, no IOMMU?!
12:25:29.901421 [init -> nvme_drv] CAP:  Mqes:2048 Cqr:1 Ams:0 To:15 Dstrd:0 Nssrs:0 Css:1 Bps:0 Mpsmin:0 Mpsmax:4
12:25:29.901750 [init -> nvme_drv] VS:   1.2.0
12:25:29.902415 [init -> nvme_drv] Warning: ignore mpsmax:4
12:25:41.566458 [init -> nvme_drv] vid:0x8086
12:25:41.567609 [init -> nvme_drv] ssvid:0x1af4
12:25:41.567635 [init -> nvme_drv] oacs:0x0
12:25:41.567641 [init -> nvme_drv]   nsm:0x0
12:25:41.567645 [init -> nvme_drv] sn:'fnord'
12:25:41.568068 [init -> nvme_drv] mn:'QEMU NVMe Ctrl'
[…]
```

notice the jump from 12:25:**29**.902415 to 12:25:**41**.566458.

Reverting commit 9f9bed7aa889 makes the driver start-up as expected:

```
[…]
12:30:49.054906 [init -> nvme_drv] VS:   1.2.0
12:30:49.055910 [init -> nvme_drv] Warning: ignore mpsmax:4
12:30:49.223347 [init -> nvme_drv] vid:0x8086
[…]
```
The commit was made to make sure that all DMA allocations are made after the device was properly acquired and all allocation make use of an potentially existing IOMMU (#4665) I instrumented those allocations in the `Platform::Dma_buffer` utility:

```
12:40:12.203193 [ 0] sys_assign_pci: Invalid Hint (0x18)
12:40:12.204107 [init -> platform_drv] Assignment of PCI device 00:03.0 to device PD failed, no IOMMU?!
12:40:12.236726 [init -> nvme_drv] Error: _alloc:33: size: 4096
12:40:12.242013 [init -> nvme_drv] Error: _alloc:35: size: 4096 cap<34369>
12:40:12.243528 [init -> nvme_drv] Error: _alloc:33: size: 4096
12:40:12.248998 [init -> nvme_drv] Error: _alloc:35: size: 4096 cap<34370>
12:40:12.251179 [init -> nvme_drv] Error: _alloc:33: size: 2097152
12:40:12.261576 [init -> nvme_drv] Error: _alloc:35: size: 2097152 cap<34371>
12:40:12.267384 [init -> nvme_drv] CAP:  Mqes:2048 Cqr:1 Ams:0 To:15 Dstrd:0 Nssrs:0 Css:1 Bps:0 Mpsmin:0 Mpsmax:4
12:40:12.267670 [init -> nvme_drv] VS:   1.2.0
12:40:12.268329 [init -> nvme_drv] Warning: ignore mpsmax:4
12:40:12.268751 [init -> nvme_drv] Error: _alloc:33: size: 2048
12:40:12.280283 [init -> nvme_drv] Error: _alloc:35: size: 2048 cap<34372>
12:40:12.282274 [init -> nvme_drv] Error: _alloc:33: size: 8192
12:40:12.287687 [init -> nvme_drv] Error: _alloc:35: size: 8192 cap<34373>
12:40:12.364421 [init -> nvme_drv] Error: _alloc:33: size: 1
12:40:23.395638 [init -> nvme_drv] Error: _alloc:35: size: 1 cap<34374>
12:40:23.472512 [init -> nvme_drv] vid:0x8086
```

Notice the suspiciously small allocation (addressed by commit cd5d0ce68) that leads to the long wait.

chelmuth, 2023-03-24:
Now, the question is: Should we address this and, if yes, how?

chelmuth, 2023-03-28:
I merged 7c8accfccc10010935fd04b9666f3a0a8f10345c to mitigate the effect just now.

nfeske, 2023-05-30:
Fixed in master.

closed 2023-05-30 by nfeske