genodelabs/genode #4795: usb: synchronization between user API and bottom-half context in dde_linux usb_host

chelmuth, 2023-03-24:
The dynamics of unplugged USB devices (esp. cascaded hubs) revealed synchronization issues that manifest in `usb_device::actconfig` unexpectedly becoming `nullptr` in the middle of sequential code paths that could be preempted. In this issue, semantics of the *USB device lock* and reuse of *drivers/usb/core/devio.c* will be elicited.

chelmuth, 2023-05-03:
@skalk Yesterday's merge of your commits revealed inconsistencies with allwinner/imx.

```
[2023-05-03 10:58:34] /usr/local/genode/tool/21.05/bin/../lib/gcc/aarch64-none-elf/10.3.0/../../../../aarch64-none-elf/bin/ld: lx_emul/usb.o: in function `kill_pid_usb_asyncio':
[2023-05-03 10:58:34] /data/genode/repos/dde_linux/src/lib/lx_emul/usb.c:840: multiple definition of `kill_pid_usb_asyncio'; generated_dummies.o:/data/genode-allwinner/src/drivers/usb_host/a64/generated_dummies.c:365: first defined here

...
[2023-05-03 10:58:34] /usr/local/genode/tool/21.05/bin/../lib/gcc/aarch64-none-elf/10.3.0/../../../../aarch64-none-elf/bin/ld: lx_emul/usb.o: in function `kill_pid_usb_asyncio':
[2023-05-03 10:58:34] /data/genode/repos/dde_linux/src/lib/lx_emul/usb.c:840: multiple definition of `kill_pid_usb_asyncio'; generated_dummies.o:/data/genode-imx/src/drivers/usb_host/imx8mq/generated_dummies.c:634: first defined here
```

chelmuth, 2023-05-03:
`make -C build/x86_64 run/smartcard KERNEL=nova` fails like follows.

```
[2023-05-03 04:52:04] [init -> test-smartcard] libusb: retrieve configuration descriptor
[2023-05-03 04:52:04] [init -> usb_drv -> usb_drv] Error: Lx_kit::Scheduler cannot find task 0x2272f8
```

skalk, 2023-05-03:
@chelmuth I've adapted imx8mq_usb_host_drv and a64_usb_host_drv to the generic lx_emul usb changes.

However, when testing the a64_usb_host_drv on the pinephone, I've encountered another structural problem in the recent work, which I've to look at tomorrow, as well as the problem shown by the smartcard test.

skalk, 2023-05-04:
@chelmuth I've prepared a fixup commit on my merge_to_staging branch, which fixes the smartcard test and pinephone usb network over modem scenario.

chelmuth, 2023-05-05:
After this issue is completed, we have to reavaluate #4682 and #4566.

closed 2023-05-30 by chelmuth