genodelabs/genode #4796: hw: scheduling oddities

cproc, 2023-03-27:
Commit 3f1a063 contains a few debugging tests which show some oddities with hw scheduling compared to nova scheduling and also compared to hw scheduling with the same activities but executed by different threads or different components (tested on x86_64 (nova, hw) and arm_v8a (hw)). The first two differences of execution time percentages according to top output (in bold text in the following table) look as if with hw the ep thread gets more execution time than other threads of the same component and the third difference looks as if with hw a time-burning thread gets much more execution time than threads with high lock contention.

| | ep| thread 1| thread 2 | thread 3 | ep (component 2) | ep (component 3) |
|---|---|---|---|---|---|---|
| | | | | | | |
| **activity** | burn | | | | burn | |
| **nova** | ~50% | | | | ~50% | |
| **hw** | ~50% | | | | ~50% | |
| | | | | | | |
| **activity** | sleep | burn | burn | | |
| **nova** | | ~50% | ~50% | | | |
| **hw** | | ~50% | ~50% | | | |
| | | | | | | |
| **activity** | burn | burn| | | | |
| **nova** | ~50% | ~50% | | | | |
| **hw** | **~66%** | **~33%** | | | | |
| | | | | | | |
| **activity** | burn | | | | burn | burn |
| **nova** | ~33% | | | | ~33% | ~33% |
| **hw** | ~33% | | | | ~33% | ~33% |
| | | | | | | |
| **activity** | sleep | burn | burn | burn | | |
| **nova** | | ~33% | ~33% | ~33% | | |
| **hw** | | ~33% | ~33% | ~33% | | |
| | | | | | | |
| **activity** | burn | burn| burn | | | |
| **nova** | ~33% | ~33% | ~33% | | | |
| **hw** | **~50%** | **~25%** | **~25%** | | | |
| | | | | | | |
| **activity** | sleep | burn | mutex acquire/release| mutex acquire/release| | |
| **nova** | | ~40% | ~30% | ~30% | | |
| **hw** | | **~99.7%** | **~0.08%** | **~0.08%** | | |

m-stein, 2023-03-27:
Commit https://github.com/m-stein/genode/commit/37415083145301eae4f580e6b58246540935bf7c mitigates the effect in the 2mutex/1burner situation. The distribution of execution time is now 50%/25%/25%. I don't know yet why the commit doesn't produce something like 33%/33%/33% but will go on investigating. Does this commit raise performance in your scenario?

chelmuth, 2023-03-28:
> The distribution of execution time is now 50%/25%/25%. I don't know yet why the commit doesn't produce something like 33%/33%/33% but will go on investigating.

I may be wrong but isn't this to be expected if thread1 and thread2 run mutually-exclusive for significant time intervals but burner is runnable all the time?

cproc, 2023-03-29:
> Does this commit raise performance in your scenario?

The audio dropouts still exist, but it looks like the maximum wait time for a timer lock is now mostly below 50 ms compared to over 80 ms or sometimes over 100 ms without the commit!

ssumpf, 2023-04-19:
7b95b68 contains a small `signal_stress.run` script that spawns a test that causes a signal storm, while also calling `wait_and_dispatch_one_io_signal` in a loop. Next to it the script starts a CPU burner with 100% utilization as well as top.

Results for base-hw:

1. Master:
```
cpu=0.0   3     0  49.92%   0.00% thread='burn_0x0'                label='init -> cpu_burner.1'
cpu=0.0   3     0  49.91%   0.00% thread='sender_thread'           label='init -> test-signal_loop'
cpu=0.0   0     0   0.09%   0.00% thread='entrypoint'              label='core'
cpu=0.0   3     0   0.05%   0.00% thread='ep'                      label='init -> test-signal_loop'
cpu=0.0   3   163   0.00%   0.00% thread='top'                     label='init -> top'
cpu=0.0   3     0   0.00%   0.00% thread='ep'                      label='init -> timer'
```
2. With commit 3138612:
```
cpu=0.0   3     0  99.34%   0.00% thread='sender_thread'           label='init -> test-signal_loop'
cpu=0.0   3     0   0.28%   0.00% thread='burn_0x0'                label='init -> cpu_burner.1'
cpu=0.0   3     0   0.23%   0.00% thread='ep'                      label='init -> test-signal_loop'
cpu=0.0   3   163   0.13%   0.00% thread='ep'                      label='init -> top'             
cpu=0.0   3     0   0.00%   0.00% thread='timer'                   label='init -> timer'           
cpu=0.0   3   163   0.00%   0.00% thread='top'                     label='init -> top'       
```
3. With commit 3138612 and 6735426 (that moves helping shares to the tail of the fills list)
```
cpu=0.0   3     0  66.79%   0.00% thread='sender_thread'           label='init -> test-signal_loop'
cpu=0.0   3     0  33.00%   0.00% thread='burn_0x0'                label='init -> cpu_burner.1'    
cpu=0.0   3     0   0.07%   0.00% thread='ep'                      label='init -> test-signal_loop'
cpu=0.0   0     0   0.05%   0.00% thread='entrypoint'              label='core'                    
cpu=0.0   3   163   0.03%   0.00% thread='ep'                      label='init -> top'       
cpu=0.0   3     0   0.00%   0.00% thread='ep'                      label='init -> cpu_burner.1'    
```
Clearly, 1. would be the desired behavior.

skalk, 2024-04-16:
Commit 36d56c6 fixes the oddity @cproc has analyzed so thoroughly with regard to mutex picking threads, and a computation intensive one. The other different behaviour regarding the first EP thread vs. another thread started beyond, comes from the fact that under base-hw the main thread, as well as the EP thread both have their own timeslice (in contrast to Nova), and the main thread which synchronizes with the EP at startup sends an IPC. As long as the EP does not respond to this IPC, it runs with this additional timeslice by the helping mechanism. So this difference is intended behaviour as already admirable analyzed by @chelmuth and others.
Therefore, I mark this issue as fixed.

nfeske, 2024-04-16:
Merged to staging. A quick test running Sculpt 24.04 using base-hw on my x250 works nicely.

chelmuth, 2024-04-18:
Commit fdaf512c7363ac23662ae91a675184bcd0c29ce1 significantly increases handler activations in the test-signal throughput test (depot_autopilot), which honors the goal of minimized latency. The following numbers were measured on Fujitsu U7411 (i7-1185G7).

```
hw (master)

[init -> test -> test-signal] sender submitted a total of 13902765 signals
[init -> test -> test-signal] handler received a total of 13902765 signals
[init -> test -> test-signal] handler was activated 504 times (27584 signals per activation)

hw (staging)

[init -> test -> test-signal] sender submitted a total of 3945946 signals
[init -> test -> test-signal] handler received a total of 3991494 signals
[init -> test -> test-signal] handler was activated 1198982 times (3 signals per activation)

nova

[init -> test -> test-signal] sender submitted a total of 62634552 signals
[init -> test -> test-signal] handler received a total of 62634552 signals
[init -> test -> test-signal] handler was activated 505 times (124028 signals per activation)
```

Please note that the test fails on _hw (staging)_ because 3945946 submitted signals led to the reception of 3991494 signals, which is a plus of 45548. @skalk any ideas? The test uses `Signal::num()`.

chelmuth, 2024-04-24:
Reopen as `test-signal` still fails like described.

skalk, 2024-04-24:
The changed scheduling order uncovered this wrong counting of signals in the test. I am 100% sure it is not related to the scheduling change itself. I'll open another issue for this.

nfeske, 2024-04-26:
Fixed in master.

closed 2024-04-26 by nfeske