genodelabs/genode #4802: Please add `tool/depot/list_archives` to list available"API/source/package/raw archives" for `tool/depot/create`

l-johanson, 2023-04-03:
Dear genode-team,

Please add a CLI tool called e.g. `tool/depot/list_archives` to list "API/source/package/raw/bin archives" for `tool/depot/create`.

In order to populate my depot, I use `tool/depot/create` like this:
`PREPARE_PORTS=1 UPDATE_VERSIONS=1 tool/depot/create l-johanson/bin/x86_64/which`

As a user, who wants to build all packages,
I'd like to have an enumeration of all available "API/source/package/raw archives",
to build them locally / have them offline available in my depot.

1. So I currently use this workaround, to list all available  "API/source/raw archives":
`find . -name "content.mk" | awk -F"/" '{print $5"/"$6}'`
2. The following to list all available "bin archives":
`find . -name "content.mk" | awk -F"/" '{print $5"/"$6}' | grep '^src' | sed 's/^src/bin\/x86_64/g'`
3. And the following to list all available "pkg archives":
`find repos/ -name "archives" | awk -F"/" '{print $4"/"$5}' | sed 's/^pkg/pkg\/x86_64/g'`

With having `tool/depot/list_archives`, I'd like to build all packages with:
`PREPARE_PORTS=1 UPDATE_VERSIONS=1 tool/depot/create $(tool/depot/list_archives)`
So ideally, an environment var like `DEPOT_USER=l-johanson` would prepend this string before all archives enumerated by `tool/depot/list_archives`, if specified.

Please tell me, if should go about implementing this.

Kind regards,
Lars

P.S.: I used a meta-pkg to build all archives before, but the above stated option seems cleaner to me. Also it seems more informative to other user as well, who would like to know, what archives are available for build and usage.

m-stein, 2023-04-18:
@l-johanson I personally never had the use case you're describing and I expect it to be rather rare that a user wants to create all available archives. For the more regular case when I want to build an individual collection of archives (normally packages), I create a pseudo-package. That said, I don't want to stop you from implementing such a tool for your personal use but I have my doubts that it would be merged to mainline.

chelmuth, 2023-08-24:
As we'll not address this issue in the near future and the OP did not respond, I close it.

closed 2023-08-24 by chelmuth