genodelabs/genode #4803: Attaching to sub rm leads to cap quota limit exceeded error

S7rizh, 2023-04-03:
Hello! 

If a region map is attached to a component's RM, performing more than a certain amount of "attach" operations on a region map lead to the `Genode::Quota_guard<Genode::Cap_quota>::Limit_exceeded` error. The error can be reproduced by applying the following patch to the `repos/base/src/test/sub_rm` test. But I am not sure if it is an error or intended behavior.

```
diff --git a/repos/base/src/test/sub_rm/main.cc b/repos/base/src/test/sub_rm/main.cc
index 1dad1106d1..36bf7544e1 100644
--- a/repos/base/src/test/sub_rm/main.cc
+++ b/repos/base/src/test/sub_rm/main.cc
@@ -62,7 +62,7 @@ void Component::construct(Env &env)
        log("--- sub-rm test ---");

        log("create RM connection");
-       enum { SUB_RM_SIZE = 1024*1024 };
+       enum { SUB_RM_SIZE = 2*1024*1024 };
        Rm_connection rm(env);

        /*
@@ -191,6 +191,15 @@ void Component::construct(Env &env)
        sub_rm.attach_at(ds, DS_SUB_OFFSET_4, 2*4096, 4096);
        validate_pattern_at(test_pattern_2(), sub_rm_base + DS_SUB_OFFSET_4);

+       /*
+        * Stress test attaching 300 pages
+       */
+       for (unsigned long i=0; i<300; i++){
+               const unsigned long offset = DS_SUB_OFFSET_4 + DS_SIZE;
+               log("attaching to sub rm at ",  offset +  i *4096, " [", i, "]");
+               sub_rm.attach_at(ds, offset + i * 4096, 4096, 0);
+       }
+
        /*
         * Detach the first attachment (to be validated by the run script by
         * inspecting '/proc/pid/maps' after running the test.
```

And the output on x86_64 nova looks as follows.

```
[init -> test-sub_rm] attaching to sub rm at 1196032 [96]
[init -> test-sub_rm] attaching to sub rm at 1200128 [97]
[init -> test-sub_rm] attaching to sub rm at 1204224 [98]
[init -> test-sub_rm] attaching to sub rm at 1208320 [99]
[init -> test-sub_rm] attaching to sub rm at 1212416 [100]
[init -> test-sub_rm] attaching to sub rm at 1216512 [101]
[init -> test-sub_rm] attaching to sub rm at 1220608 [102]
[init -> test-sub_rm] attaching to sub rm at 1224704 [103]
[init -> test-sub_rm] attaching to sub rm at 1228800 [104]
[init -> test-sub_rm] Error: Uncaught exception of type 'Genode::Quota_guard<Genode::Cap_quota>::Limit_exceeded'
[init -> test-sub_rm] Warning: abort called - thread: ep
```

Also:
- I was able to reproduce this error on hw and linux platforms. However, on linux, it outputs a `Genode::Native_capability::Reference_count_overflow` error. 
- Same error occurs if attaching to root rm on linux.
- Same error occurs if more than one dataspace is being attached. It fails after the same number of attach operations.

nfeske, 2023-04-04:
The cap-quota-exceeded exception must be expected when calling `attach_at`. See the documentation of `attach` (https://github.com/genodelabs/genode/blob/master/repos/base/include/region_map/region_map.h#L94) for the various exception cases.

The most appropriate response for a depleted caps or RAM quota would be to `upgrade` the `rm` connection (see https://github.com/genodelabs/genode/blob/master/repos/base/include/base/connection.h#L49). This is a common pattern, e.g., as found at https://github.com/genodelabs/genode/blob/master/repos/base/include/rm_session/connection.h#L35.

The `Reference_count_overflow` exception on Linux is not immediately clear to me. On Linux, the RM service is provided locally from within the component (you may grep the base-linux code for `region_map_mmap`). I haven't checked the code, but should this implementation hold capability copies as meta data (like one copy per attached region), such a ref-count overflow may occur because the test code creates more than 255 regions using the same dataspace capability.

nfeske, 2023-11-30:
I'm closing this issue as it has become stale.

closed 2023-11-30 by nfeske