genodelabs/genode #4807: wifi: use latest available firmware

cnuke, 2023-04-06:


cnuke, 2023-04-06:
Commt [166c2de](https://github.com/cnuke/genode/commit/166c2de4ce7fab3f5e28fd223c15a73f86e6ec37) cleans up the used firmware images in the `pc_wifi_drv` for iwlwifi devices.

Since I have not all devices at hand to test the changes I would appreciate if 9xxx series owners (@a-dmg, @jschlatow, @m-stein) could give the branch ([issue#4807-wifi-firmware-update-2023-04-06](https://github.com/cnuke/genode/compare/issue%234807-wifi-firmware-update-2023-04-06)) a try.

a-dmg, 2023-04-07:
@cnuke I have tested on my blade 15, and this works fine on sculpt 👍🏻

cnuke, 2023-04-11:
@a-dmg thanks for testing!

(I am going to add a *provisional* fixed* label as I don't expect other issues.)

jschlatow, 2023-04-11:
@cnuke I just tested the your changes on the T490s. The wifi_drv appears to operate normally, however, I noticed a bunch of errors in the log:

```
[runtime -> wifi_drv] [34mWarning: ignoring accesspoint with invalid ssid[0m
[runtime -> wifi_drv] NR_IRQS: 4352, nr_irqs: 4352, preallocated irqs: 256[0m
[runtime -> wifi_drv] clocksource: dde_counter: mask: 0xffffffffffffff max_cycles: 0x1d854df40, max_idle_ns: 3526361616960 ns[0m
[runtime -> wifi_drv] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 7645041785100000 ns[0m
[runtime -> wifi_drv] NET: Registered PF_NETLINK/PF_ROUTE protocol family[0m
[runtime -> wifi_drv] clocksource: Switched to clocksource dde_counter[0m
[runtime -> wifi_drv] Intel(R) Wireless WiFi driver for Linux[0m
[runtime -> wifi_drv] iwlwifi 00:14.3: pci_enable_msi failed - -38[0m
[runtime -> wifi_drv] task_work_add: task: 0000000088fe2821 work: 0000000095370869 notify: 0[0m
[runtime -> wifi_drv] [31mError: lx_emul_request_firmware_nowait: name: 'iwlwifi-9000-pu-b0-jf-b0-46.ucode'[0m
[runtime -> wifi_drv] [31mError: lx_emul_request_firmware_nowait: fw_name: 'iwlwifi-9000-pu-b0-jf-b0-46.ucode'[0m
[runtime -> wifi_drv] iwlwifi 00:14.3: WRT: Overriding region id 0[0m
[runtime -> wifi_drv] iwlwifi 00:14.3: WRT: Overriding region id 1[0m
[runtime -> wifi_drv] iwlwifi 00:14.3: WRT: Overriding region id 2[0m
[runtime -> wifi_drv] iwlwifi 00:14.3: WRT: Overriding region id 3[0m
[runtime -> wifi_drv] iwlwifi 00:14.3: WRT: Overriding region id 4[0m
[runtime -> wifi_drv] iwlwifi 00:14.3: WRT: Overriding region id 6[0m
[runtime -> wifi_drv] iwlwifi 00:14.3: WRT: Overriding region id 8[0m
[runtime -> wifi_drv] iwlwifi 00:14.3: WRT: Overriding region id 9[0m
[runtime -> wifi_drv] iwlwifi 00:14.3: WRT: Overriding region id 10[0m
[runtime -> wifi_drv] iwlwifi 00:14.3: WRT: Overriding region id 11[0m
[runtime -> wifi_drv] iwlwifi 00:14.3: WRT: Overriding region id 15[0m
[runtime -> wifi_drv] iwlwifi 00:14.3: WRT: Overriding region id 16[0m
[runtime -> wifi_drv] iwlwifi 00:14.3: WRT: Overriding region id 18[0m
[runtime -> wifi_drv] iwlwifi 00:14.3: WRT: Overriding region id 19[0m
[runtime -> wifi_drv] iwlwifi 00:14.3: WRT: Overriding region id 20[0m
[runtime -> wifi_drv] iwlwifi 00:14.3: WRT: Overriding region id 21[0m
[runtime -> wifi_drv] iwlwifi 00:14.3: WRT: Overriding region id 28[0m
[runtime -> wifi_drv] [31mError: lx_emul_request_firmware_nowait: name: 'iwl-debug-yoyo.bin'[0m
[runtime -> wifi_drv] iwlwifi 00:14.3: loaded firmware version 46.6b541b68.0 9000-pu-b0-jf-b0-46.ucode op_mode iwlmvm[0m
[runtime -> wifi_drv] iwlwifi 00:14.3: Detected Intel(R) Wireless-AC 9560 160MHz, REV=0x312[0m
[runtime -> wifi_drv] iwlwifi 00:14.3: base HW address: 04:33:c2:1d:af:1f, OTP minor version: 0x4[0m
[runtime -> wifi_drv] ieee80211 phy0: Selected rate control algorithm 'iwl-mvm-rs'[0m
[runtime -> wifi_drv] NET: Registered PF_PACKET protocol family[0m
[runtime -> wifi_drv] sched_clock: Marking stable (164179000, 14464000)->(186625000, -7982000)[0m
[runtime -> wifi_drv] reg: Loading compiled-in X.509 certificates for regulatory database[0m
[runtime -> wifi_drv] [31mError: lx_emul_request_firmware_nowait: name: 'regulatory.db'[0m
[runtime -> wifi_drv] [31mError: lx_emul_request_firmware_nowait: fw_name: 'regulatory.db'[0m
[runtime -> wifi_drv] [31mError: lx_emul_request_firmware_nowait: name: 'regulatory.db.p7s'[0m
[runtime -> wifi_drv] [31mError: lx_emul_request_firmware_nowait: fw_name: 'regulatory.db.p7s'[0m
[runtime -> wifi_drv] reg: loaded regulatory.db is malformed or signature is missing/invalid[0m
[runtime -> wifi_drv] [34mWarning: /dev/rtc not readable, returning 0[0m
[runtime -> wifi_drv] [34mWarning: Libc RNG not configured[0m
[drivers -> usb_drv] usb 3-7: USB disconnect, device number 4[0m
[drivers -> usb_drv] usb 3-7: new high-speed USB device number 10 using xhci_hcd[0m
[drivers -> usb_drv] usb 3-7: USB disconnect, device number 10[0m
[drivers -> usb_drv] usb 3-7: new high-speed USB device number 11 using xhci_hcd[0m
[runtime -> wifi_drv] wlan0: authenticate with 74:42:7f:93:66:10[0m
[runtime -> wifi_drv] wlan0: 80 MHz not supported, disabling VHT[0m
[runtime -> wifi_drv] wlan0: send auth to 74:42:7f:93:66:10 (try 1/3)[0m
[runtime -> wifi_drv] wlan0: authenticated[0m
[runtime -> wifi_drv] wlan0: associate with 74:42:7f:93:66:10 (try 1/3)[0m
[runtime -> wifi_drv] wlan0: RX AssocResp from 74:42:7f:93:66:10 (capab=0x1431 status=0 aid=1)[0m
[runtime -> wifi_drv] wlan0: associated[0m
[runtime -> wifi_drv] create uplink for net device wlan0[0m
[runtime -> nic_router] [uplink] NIC sessions: 1[0m
[runtime -> wifi_drv] wlan0: disassociated from 74:42:7f:93:66:10 (Reason: 2=PREV_AUTH_NOT_VALID)[0m
[runtime -> nic_router] [uplink] dynamic IP config: none[0m
[runtime -> nic_router] [uplink] NIC sessions: 0[0m
[drivers -> usb_drv] usb 3-7: USB disconnect, device number 11[0m
[drivers -> usb_drv] usb 3-7: new high-speed USB device number 12 using xhci_hcd[0m
[drivers -> usb_drv] usb 3-7: USB disconnect, device number 12[0m
[drivers -> usb_drv] usb 3-7: new high-speed USB device number 13 using xhci_hcd[0m
[runtime -> wifi_drv] [34mWarning: deferring config update (fill network pass)[0m
[runtime -> wifi_drv] wlan0: authenticate with 74:42:7f:93:66:10[0m
[runtime -> wifi_drv] wlan0: 80 MHz not supported, disabling VHT[0m
[runtime -> wifi_drv] wlan0: send auth to 74:42:7f:93:66:10 (try 1/3)[0m
[runtime -> wifi_drv] wlan0: authenticated[0m
[runtime -> wifi_drv] wlan0: associate with 74:42:7f:93:66:10 (try 1/3)[0m
[runtime -> wifi_drv] wlan0: RX AssocResp from 74:42:7f:93:66:10 (capab=0x1431 status=0 aid=1)[0m
[runtime -> wifi_drv] wlan0: associated[0m
[runtime -> wifi_drv] create uplink for net device wlan0[0m
[runtime -> nic_router] [uplink] NIC sessions: 1[0m

```

cnuke, 2023-04-11:
@jschlatow thanks for testing - if your refer to the `lx_emul_request_firmware_nowait` error message those stem from the *DBG* commit and can be safely ignored.

commit 166c2de4 (2023-04-11): wifi: use latest available firmware

This commit updates the used firmware images for the iwlwifi device
to latest available ones (supported by the current 6.1.x driver).
It also removes old firmware images that by now are not used anymore.

Fixes #4807.
closed 2023-04-17 by chelmuth