genodelabs/genode #4809: DDE Linux struct page object aliasing

cnuke, 2023-04-13:
Due to the way the memory-allocation code is currently implemented and the `struct page` object handling is integrate aliasing that has unexpected side-effects can occur. Since `struct page` objects are created for a page-size aligned address [1] when requested, allocations that fall within the same virtual page address range share the same object.

Such objects are created lazy and and life-time management is only performed via ref-counting when the page API [2] (`alloc_pages`, `free_pages` etc.) is used. Since the current implementation removes `struct page` objects [3] when performing allocations still referenced objects can become invalid and can be corrupted as the memory is re-used.

[1] https://github.com/genodelabs/genode/blob/73771669f1c3a2c01555ebc01c7741b5a37b3d51/repos/dde_linux/src/lib/lx_emul/virt_to_page.c#L20-L37

[2] [shadow/mmc/page_alloc.c](https://github.com/genodelabs/genode/blob/73771669f1c3a2c01555ebc01c7741b5a37b3d51/repos/dde_linux/src/lib/lx_emul/shadow/mm/page_alloc.c)

[3] https://github.com/genodelabs/genode/blob/73771669f1c3a2c01555ebc01c7741b5a37b3d51/repos/dde_linux/src/lib/lx_emul/alloc.cc#L20-L25

chelmuth, 2023-12-01:
The real fix is still not merged.

chelmuth, 2023-12-05:
Commit cfe7915573a5a7c998c07f7c346b456dfee46b21 implements page-struct management per DMA buffer. Repositories for allwinner, imx, rpi, and zynq were also adapted for the change. NOw I cross fingers for nightly. Also, any test results for Pinephone and MNT Reform are much appreciated.

cnuke, 2023-12-05:
I briefly tested the _glmark2_ preset on the PinePhone with the current staging branch. It works with the page-struct commits but after the sixth scenario the gpu driver fails to upgrade the `Platform` session and issues a RAM resource request of `16M`. When I revert [cfe7915](https://github.com/genodelabs/genode/commit/cfe7915573a5a7c998c07f7c346b456dfee46b21) and [06b00b5](https://github.com/genodelabs/genode-allwinner/commit/06b00b506c914475123b7b1ea87edbe3bc555b77) it finishes successfully without any resource request.

I can look into that later this week.

commit cfe79155 (2023-12-05): lx_emul: manage page structs per buffer range

The management of Linux page structs is now tied to the life time of DMA
buffers. Thus, page structs are created when a buffer is allocated and
deallocated only when a buffer is freed - not on lx_emul_mem_free()
because DMA buffers are cached. Page struct refcounting was entirely
reworked in lx_emul/shadow/mm/page_alloc.c.

Fixes #4809
commit 1e7116fc (2023-12-13): lx_emul: manage page structs per buffer range

The management of Linux page structs is now tied to the life time of DMA
buffers. Thus, page structs are created when a buffer is allocated and
deallocated only when a buffer is freed - not on lx_emul_mem_free()
because DMA buffers are cached. Page struct refcounting was entirely
reworked in lx_emul/shadow/mm/page_alloc.c.

Fixes #4809
closed 2023-12-13 by nfeske