genodelabs/genode #4810: vfs_lxip: sendto(UDP) somtimes uses wrong destination address

chelmuth, 2023-04-14:
With the current socket FS, `sendto()` is implemented by two consecutive writes - first to the *remote* node and second to the *data* node of the socket. Hence, the order of the write operations is important, but the libc socket code does not ensure this order (e.g., by sync or read). In the VFS server, operations on *distinct* nodes expectably do not preserve the order of operations in the packet stream. Therefore in some cases, the writes to remote and data get reordered in `_execute_jobs()` if the nodes are already *active* (due to read-ready notifications) in the `_active_nodes` queue but in reverse order.

chelmuth, 2024-01-05:
@ssumpf this is the issue I mentioned

nfeske, 2025-04-29:
@ssumpf may this issue have become obsolete with your recent rework of vfs_lwip/lxip?

chelmuth, 2025-04-29:
As the update did not address the two consecutive writes to the socket FS, the issue most likely still persists. Unfortunately, we never implemented an in-house test case for this issue, which was reported by a user via email.