genodelabs/genode #4812: platform_drv: allow PCI bus addressing of devices

trimpim, 2023-04-17:
We have a board with two Ethernet controllers (UP-Xtreme) both of these devices have the same `vendor_iid` and `device_id`. 

To have a predictable assignment of the devices to drivers, the policies of the `platform_drv` should support the triplet `bus`, `device` and `function`.

chelmuth, 2023-04-19:
Would you mind do document the feature in the README?

skalk, 2023-04-19:
@trimpim: if you need to distinguish in between different devices with same attributes, why don't you use the name of the device as policy item, which in case of PCI devices is the BDF tupel anyway? I would favor that in contrast to re-add XML attributes to the configuration and add additional code to the crucial policy code in the platform driver. Otherwise, please describe your use-case in more detail, why it is not sufficient to use the name.

trimpim, 2023-04-19:
@skalk I wasn't aware, that the name is actually the BDF for PCI devices. I will change this.

@chelmuth I will update the README.

trimpim, 2023-04-19:
@skalk sorry I misunderstood you comment at first. This patch is actually not needed.

@chelmuth Therefore I close the issue.

closed 2023-04-19 by trimpim