genodelabs/genode #4824: libc/libunwind: add missing symbols required by the Rust std library

atopia, 2023-04-24:
In order to reuse Rust's x86_64-unknown-freebsd target triple for Rust, a number of symbols have to be added to our standard library.
As per previous offline discussion, currently `libunwind` related functions are split out to a separate library. Given that for now only dummy implementations are supplied anyway, those could alternately be added to our `libc`, which would very slightly bloat it for regular users but spare users of the Rust runtime an extra library dependency.

chelmuth, 2023-04-27:
May these symbols be part of libgcc? https://gcc.gnu.org/git/?p=gcc.git;a=blob;f=libgcc/unwind-seh.c;h=4a6c40bf176c893b16c6651bef2f71c69bf1bde9;hb=HEAD

cproc, 2023-04-27:
> May these symbols be part of libgcc? https://gcc.gnu.org/git/?p=gcc.git;a=blob;f=libgcc/unwind-seh.c;h=4a6c40bf176c893b16c6651bef2f71c69bf1bde9;hb=HEAD

libbacktrace uses two of the `_Unwind_*` symbols from libgcc_eh.a:

https://github.com/cproc/genode-world/blob/107b0cb11bc68854241758be6b0d4d0419f08918/lib/import/import-backtrace.mk

atopia, 2023-05-03:
Thank you @chelmuth and @cproc for the helpful hints!

nfeske, 2023-05-05:
Merged to staging.

commit d8cc8356 (2023-05-05): libc: implement signal(3)

Fixes #4824
closed 2023-05-30 by chelmuth