genodelabs/genode #4826: base-hw: add support for SVM virtualization

atopia, 2023-04-28:
One of the remaining feature gaps of the _base-hw_ kernel is virtualization support on X86.

This issue serves as the preliminary tracking issue for implementing support for AMD's Secure Virtual Machine (SVM). The work on x86 virtualization support as a whole can be split across multiple tickets as needed.

skalk, 2023-05-22:
@atopia: I reviewed your latest commit series. Thanks for your careful adaptions regarding the system-call interface so far! Apart from the minor comments I made inline, the only issue left is the commit order I guess. Just taking the very first ones leads to a broken core compilation for base-hw on PC. Maybe, a simple re-ordering is enough, maybe you've to squash some of the commits together.

atopia, 2023-05-23:
@skalk Thanks for your review. Please have a look at https://github.com/atopia/genode/tree/svm-support-staging for the latest patch series that incorporates your comments. I have pushed a rebased branch of the seoul change to https://github.com/atopia/genode-world/tree/seoul-for-hw-staging as well.

skalk, 2023-05-23:
@atopia: thanks, I'm done with my review. Apart from the point that @chelmuth made regarding the kernel-agnostic Vcpu_state structures in `repos/base`, the commit series is ready for staging in my eyes.

atopia, 2023-05-24:
atopia/genode@a6f1b5f should fix the startup state issues and fixes the spelling of 'runnable'. Thanks @alex-ab for your debugging support!

chelmuth, 2023-05-26:
I merged the remaining two commits and successfully ran TinyCore Linux in seoul :clap: @atopia please have a look at the fixups I did while reviewing the code.

chelmuth, 2023-09-13:
Can we close this issue with #4968 in merge process?

closed 2023-09-13 by atopia