genodelabs/genode #4827: tool chain update for 23.05 release

cproc, 2023-05-02:


cproc, 2023-05-02:
My issue4827 branch [1] contains a first version of the update with binutils 2.40 and gcc 12.2.0.

TODO:

- RISC-V support
- ADA runtime update
- GDB update
- use C++20 by default

I'm going to create separate issues for the new errors which occurred so far when I tried to build Genode with the updated tools.

[1] https://github.com/cproc/genode/tree/issue4827

nfeske, 2023-05-02:
That's just fantastic! @cproc, you made my day.

chelmuth, 2023-05-04:
I tried to build the tool chain via `` but got the following errors.

```
gpr-env.adb:1867:16: no candidate interpretations match the actuals:
gpr-env.adb:1867:16: missing argument for parameter "Separators" in call to "CREATE" declared at g-arrspl.ads:97, instance at g-strspl.ads:39
gpr-env.adb:1867:16: missing argument for parameter "Separators" in call to "CREATE" declared at g-arrspl.ads:83, instance at g-strspl.ads:39
gpr-env.adb:1867:16: context requires function call, found procedure name
gpr-env.adb:1867:53: unmatched actual "Mode" in call
gpr-env.adb:1868:16: ambiguous call to "Append"
gpr-env.adb:1868:16: possible interpretation at a-coinve.ads:230, instance at gpr-util.ads:39
gpr-env.adb:1868:16: possible interpretation at a-coinve.ads:234, instance at gpr-util.ads:39
gpr-env.adb:1868:25: "P" is undefined
gpr-env.adb:1884:19: no selector "Prepend_Vector" for private type "Ada.Containers.Indefinite_Vectors.Vector" from instance at gpr-util.ads:39
gpr-env.adb:1886:19: no selector "Append_Vector" for private type "Ada.Containers.Indefinite_Vectors.Vector" from instance at gpr-util.ads:39
gprbuild-link.adb:762:20: no selector "Append_Vector" for private type "Ada.Containers.Indefinite_Vectors.Vector" from instance at gpr-util.ads:39
gprbuild-link.adb:793:22: no selector "Append_Vector" for private type "Ada.Containers.Indefinite_Vectors.Vector" from instance at gprbuild.ads:165
gprbuild-link.adb:796:22: no selector "Append_Vector" for private type "Ada.Containers.Indefinite_Vectors.Vector" from instance at gprbuild.ads:165
gprbuild-link.adb:879:19: no selector "Append_Vector" for private type "Ada.Containers.Indefinite_Vectors.Vector" from instance at gprbuild.ads:165
gprbuild-link.adb:3364:37: no selector "Append_Vector" for private type "Ada.Containers.Indefinite_Vectors.Vector" from instance at gprbuild.ads:165
gprbuild-link.adb:3383:19: no selector "Append_Vector" for private type "Ada.Containers.Indefinite_Vectors.Vector" from instance at gprbuild.ads:165
gpr-env.adb:1867:16: no candidate interpretations match the actuals:
gpr-env.adb:1867:16: missing argument for parameter "Separators" in call to "CREATE" declared at g-arrspl.ads:97, instance at g-strspl.ads:39
gpr-env.adb:1867:16: missing argument for parameter "Separators" in call to "CREATE" declared at g-arrspl.ads:83, instance at g-strspl.ads:39
gpr-env.adb:1867:16: context requires function call, found procedure name
gpr-env.adb:1867:53: unmatched actual "Mode" in call
gpr-env.adb:1868:16: ambiguous call to "Append"
gpr-env.adb:1868:16: possible interpretation at a-coinve.ads:230, instance at gpr-util.ads:39
gpr-env.adb:1868:16: possible interpretation at a-coinve.ads:234, instance at gpr-util.ads:39
gpr-env.adb:1868:25: "P" is undefined
gpr-env.adb:1884:19: no selector "Prepend_Vector" for private type "Ada.Containers.Indefinite_Vectors.Vector" from instance at gpr-util.ads:39
gpr-env.adb:1886:19: no selector "Append_Vector" for private type "Ada.Containers.Indefinite_Vectors.Vector" from instance at gpr-util.ads:39
gnatmake: "/var/tmp/build-23.05/bootstrap/gprbuild/gpr/src/gpr-env.adb" compilation error
gnatmake: "/var/tmp/build-23.05/bootstrap/gprbuild/src/gprbuild-link.adb" compilation error
gnatmake: "/var/tmp/build-23.05/bootstrap/gprbuild/gpr/src/gpr-util.adb" compilation error
make: *** [tool/tool_chain:472: /var/tmp/build-23.05/bootstrap/install/bin/gprbuild] Error 4
```

**Info about the Linux installation**

- Ubuntu 20.04.5 LTS
 - gcc (Ubuntu 9.4.0-1ubuntu1~20.04.1) 9.4.0
- GNATMAKE 9.4.0

cproc, 2023-05-04:
@chelmuth: can you execute binaries from `/var/tmp`? I wonder if maybe the host tools are used to build gprbuild instead of the bootstrap tools.

chelmuth, 2023-05-04:
Is the following correct or does it confirm your suspicion? Do you need more info?

```
> head /var/tmp/build-23.05/bootstrap/gprbuild/gprbuild.ali 
V "GNAT Lib v9"
P ZX
```

cproc, 2023-05-04:
> Is the following correct or does it confirm your suspicion? Do you need more info?
> 
> ```
> > head /var/tmp/build-23.05/bootstrap/gprbuild/gprbuild.ali 
> V "GNAT Lib v9"
> P ZX
> ```

In my Xubuntu 20.04 test VM it says

```
V "GNAT Lib v12"
P SS ZX
```

The `tool_chain` script prepends the bootstrap install location to the `PATH` environment variable, so normally the bootstrap tools should be used to build gprbuild. Since the execution permission of the bootstrap binaries is not the problem, maybe a bootstrap tool is missing for some reason? This is the output I get from `ls -1 /var/tmp/build-23.05/bootstrap/install/bin`:

```
addr2line
ar
as
c++
c++filt
cpp
elfedit
g++
gcc
gcc-ar
gccgo
gcc-nm
gcc-ranlib
gcov
gcov-dump
gcov-tool
gnat
gnatbind
gnatchop
gnatclean
gnatkr
gnatlink
gnatls
gnatmake
gnatname
gnatprep
gprbuild
gprclean
gprconfig
gprinstall
gprls
gprname
ld
ld.bfd
nm
objcopy
objdump
ranlib
readelf
size
strings
strip
x86_64-pc-linux-gnu-c++
x86_64-pc-linux-gnu-g++
x86_64-pc-linux-gnu-gcc
x86_64-pc-linux-gnu-gcc-12.2.0
x86_64-pc-linux-gnu-gcc-ar
x86_64-pc-linux-gnu-gccgo
x86_64-pc-linux-gnu-gcc-nm
x86_64-pc-linux-gnu-gcc-ranlib
```

chelmuth, 2023-05-04:
Could it be I have an incomplete/inconsistent state as I started with `ENABLE_FEATURES="c c++"` and later extended to `ENABLE_FEATURES="c c++ ada"`?

cproc, 2023-05-04:
> Could it be I have an incomplete/inconsistent state as I started with `ENABLE_FEATURES="c c++"` and later extended to `ENABLE_FEATURES="c c++ ada"`?

That could be the reason. There is currently only a make dependency on the existence of `g++` in the bootstrap install location.

chelmuth, 2023-05-04:
I deleted */var/tmp/build-23.05/bootstrap* and built from scratch without errors. Hooray :partying_face:

cproc, 2023-05-08:
GDB is now updated to version 13.1 and the RISC-V tool chain is built with `--with-arch=rv64imac_zicsr_zifencei`.

chelmuth, 2023-05-08:
Building gdb requires *libgmp-dev* on Ubuntu 20.04. *mpfr* and others seem optional.

chelmuth, 2023-05-08:
Running `make -C build/x86_64 run/log KERNEL=linux BOARD=linux` yields some deterring warnings.

```
/usr/local/genode/tool/23.05/bin/../lib/gcc/x86_64-pc-elf/12.2.0/../../../../x86_64-pc-elf/bin/ld: warning: init_main_thread.o: missing .note.GNU-stack section implies executable stack
/usr/local/genode/tool/23.05/bin/../lib/gcc/x86_64-pc-elf/12.2.0/../../../../x86_64-pc-elf/bin/ld: NOTE: This behaviour is deprecated and will be removed in a future version of the linker
/usr/local/genode/tool/23.05/bin/genode-x86-ld: warning: test.o: missing .note.GNU-stack section implies executable stack
/usr/local/genode/tool/23.05/bin/genode-x86-ld: NOTE: This behaviour is deprecated and will be removed in a future version of the linker
```

Could we prevent this?

cproc, 2023-05-08:
> Building gdb requires _libgmp-dev_ on Ubuntu 20.04. _mpfr_ and others seem optional.

We could check with pkg-config on Ubuntu 20.04, but Ubuntu 18.04 does not have pkg-config support for libgmp. Is Ubuntu 18.04 support required?

chelmuth, 2023-05-09:
> We could check with pkg-config on Ubuntu 20.04, but Ubuntu 18.04 does not have pkg-config support for libgmp. Is Ubuntu 18.04 support required?

In my opinion, we can drop 18.04 support with the release.

cproc, 2023-05-09:
Fixup commit 3331dd4 adds the libgmp check to the tool_chain script.

cproc, 2023-05-09:
I updated GCC to version 12.3.0 on my issue4827 branch.

chelmuth, 2023-05-09:
> I updated GCC to version 12.3.0 on my issue4827 branch.

Great, the server is already busy to build the new tool chain version ;-)

ssumpf, 2023-05-09:
[1a2b5c2](https://github.com/ssumpf/genode-world/commit/1a2b5c29637b99bf008f9d33d15a341e3dc3af44) fixes JDK on my `gcc-12` branch

ssumpf, 2023-05-10:
987c821 and 5164038 fix stdcxx and libc for RISC-V.

chelmuth, 2023-05-10:
> [1a2b5c2](https://github.com/ssumpf/genode-world/commit/1a2b5c29637b99bf008f9d33d15a341e3dc3af44) fixes JDK on my `gcc-12` branch
> [987c821](https://github.com/genodelabs/genode/commit/987c821da84d321d6c6f7710120d1d55a50492f4) and [5164038](https://github.com/genodelabs/genode/commit/516403825f97d1b621f6f3704edbf13d8583095d) fix stdcxx and libc for RISC-V.

Merged to staging (and updated ports manually) to let the builder chew on it.

chelmuth, 2023-05-10:
@ssumpf I also merged 5b78d276e7b9b76f095c41f8ea4deffe814dab94 en-passant as this approach looks reasonable to me. Is this okay as the commit effectively overwrite `ARCH_DMA_MINALIGN` for ARM too (where it is 128 in most cases)?

ssumpf, 2023-05-10:
> @ssumpf I also merged [5b78d27](https://github.com/genodelabs/genode/commit/5b78d276e7b9b76f095c41f8ea4deffe814dab94) en-passant as this approach looks reasonable to me. Is this okay as the commit effectively overwrite `ARCH_DMA_MINALIGN` for ARM too (where it is 128 in most cases)?

@chelmuth: Not sure, that's why I wrote to staff yesterday. We may also disable SIMD/SEE altogether. As far a I can tell it's not enabled at least for Linux x86.

cnuke, 2023-05-10:
Rather than tinkering with the `*_ALIGN` settings I think it would be best to check the used flags by Linux and add the various `'-no-*` and so on flags to the corresponding `import-*lx_emul.mk` files. That brings us closer to the original compilation.

chelmuth, 2023-05-10:
I fixed *phone_manager.prg* with genodelabs/genode-allwinner@76570433c56e6b2899a0d9b4033e4a7fb58eaf67.

chelmuth, 2023-05-10:
> Rather than tinkering with the `*_ALIGN` settings I think it would be best to check the used flags by Linux and add the various `'-no-*` and so on flags to the corresponding `import-*lx_emul.mk` files. That brings us closer to the original compilation.

Do I understand correctly that you suggest to inherit compiler options from Linux kernel, e.g. *arch/x86/Makefile* which almost expresses *-mgeneral-regs-only* by the following?

```make
KBUILD_CFLAGS += -mno-sse -mno-mmx -mno-sse2 -mno-3dnow -mno-avx
```

chelmuth, 2023-05-10:
Fixed zybo_gpio_demo_rgb.prg with genodelabs/genode-zynq@95cc6114b894b111ce3cbad83ba3284e0448785e.

chelmuth, 2023-05-10:
> yes

Looking for to your commits.

skalk, 2023-05-10:
@chelmuth commit 8d2be08 enables -mstrict-align for ARMv8 and disables FPU for ARMv7 and x86 to prevent the runtime alignment issues seen for example on i.MX 8MQ QSB. Please, give me a sign if you decide to apply it. I've to apply corresponding commits for genode-imx, genode-allwinner, genode-rpi, and genode-zynq, which depend on this one.

chelmuth, 2023-05-11:
genodelabs/genode-imx@63bd0bbf08ec988658ed5e9d49218acc501c3ae7 fixes all *error: call to ‘__bad_copy_from/to’`* errors.

cnuke, 2023-05-11:
I noticed `ARCH_DMA_MINALIGN` redefinition warnings during PinePhone testing and moved them into the _pc_ repo ([c8a48bc](https://github.com/cnuke/genode/commit/c8a48bc233f7ce317bb8a83e7fe6647e1553c7c0)). It did not appear to break anything (running Sculpt on the Fuji6) an the PC-sde of things.

cnuke, 2023-05-17:
@chelmuth please consider [af04d22](https://github.com/cnuke/genode/commit/af04d2241b18d91dff6f8d459556b972366415a1) for staging (addresses wireguard problems).

chelmuth, 2023-05-17:
> @chelmuth please consider [atopia/genode-world@cc08750](https://github.com/atopia/genode-world/commit/cc08750) for staging

Thanks, it's in.

cproc, 2023-05-22:
Thinking a bit more about the library name pattern support, `.abi.so` should actually not be included because if such a library was found, the `.abi.so` name would end up in the binary, right? So symlinks to '.lib.so' always need to be created or `ld` would need to automatically rename `.abi.so` to `.lib.so`.

cproc, 2023-05-22:
Commit 59a5162 has the `.abi.so` pattern removed.

chelmuth, 2023-05-22:
I merged the binutils-grep series just now.

cproc, 2023-05-23:
> Will still get the following build warnings. any ideas how to prevent this?

I found this blog post which has some general explanations and hints regarding these warnings:

https://www.redhat.com/en/blog/linkers-warnings-about-executable-stacks-and-segments

chelmuth, 2023-05-23:
Commit 83556dabae6eb2e09a30e1e518a2e7c091af2bff addresses the *missing .note.GNU-stack section* linker warning.

m-stein, 2023-05-25:
@cproc @chelmuth I encounter problems with vim-minimal that originate from b11c08d8237bdcec66e07697175a82e37210a7b5 that disappear when I revert the dummy_libs and ncurses stuff in vim/target.inc (reverting the LDFLAGS is not necessary, tested using a clean depot):

```
./tool/depot/create local/bin/x86_64/vim-minimal UPDATE_VERSIONS=1 FORCE=1 REBUILD= -j8 CCACHE=
created local/api/posix/2020-05-17
created local/api/noux/2023-02-26
created local/api/ncurses/2023-05-25 (new version)
created local/api/libc/2023-05-25 (new version)
created local/src/vim-minimal/2023-05-25 (new version)
checking library dependencies...
  Library ldso_so_support
  Library noux
    MERGE    noux.lib.a
    COMPILE  so_support.o
    MERGE    ldso_so_support.lib.a
  Library posix
  Library libm
    CONVERT  libm.symbols.s
  Library ncurses
  Library libc
    CONVERT  posix.symbols.s
    CONVERT  ncurses.symbols.s
    CONVERT  libc.symbols.s
    ASSEMBLE posix.symbols.o
    ASSEMBLE libm.symbols.o
    ASSEMBLE ncurses.symbols.o
    MERGE    posix.abi.so
    MERGE    libm.abi.so
    ASSEMBLE libc.symbols.o
    MERGE    ncurses.abi.so
    MERGE    libc.abi.so
  Program noux-pkg/vim-minimal/vim-minimal
    CONFIG   vim-minimal
Error: configure script of vim-minimal failed
configure: error: FAILED
make[6]: *** [/media/lypo/138e8670-1dd2-11b2-88a4-3bab3642e0c11/depot/local/api/noux/2023-02-26/mk/gnu_build.mk:160: Makefile] Error 1
make[5]: *** [var/libdeps:139: vim-minimal.prg] Error 2
make[4]: *** [Makefile:336: gen_deps_and_build_targets] Error 2
make[3]: *** [/home/lypo/genodelabs/genode/tool/depot/mk/build_bin_archive:208: /media/lypo/138e8670-1dd2-11b2-88a4-3bab3642e0c11/depot//local/bin/x86_64/vim-minimal/2023-05-25.build/bin] Error 1
make[2]: *** [/media/lypo/138e8670-1dd2-11b2-88a4-3bab3642e0c11/depot//var/build.mk:12: local/bin/x86_64/vim-minimal/2023-05-25] Error 2
make[1]: *** [/home/lypo/genodelabs/genode/tool/depot/build:136: execute_generated_build_mk_file] Error 2
make: *** [tool/depot/create:59: build] Error 2
```

m-stein, 2023-05-25:
@cproc  Sorry for the inconvenience, @chelmuth pointed me to the already existing fix.

nfeske, 2023-05-30:
Fixed in master.

closed 2023-05-30 by nfeske