genodelabs/genode #4844: acpi_drv: build error with gcc 12

cproc, 2023-05-02:
When building `drivers/acpi` with gcc 12, the following error occurs:

```
/.../repos/os/src/drivers/acpi/spec/x86/../../acpi.cc: In static member function 'static void Element::clean_list(Genode::Allocator&)':
/.../repos/os/src/drivers/acpi/spec/x86/../../acpi.cc:1241:62: error: comparing the result of pointer addition '(((char*)(& element->Element::_name)) + 128)' and NULL [-Werror=address]
 1241 |                                 freed_up += sizeof(*element) + element->_name ? element->_name_len : 0;
      |                                             ~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~
```

When I changed this line to

```
freed_up += sizeof(*element) + (element->_name ? element->_name_len : 0);
```

the following error occurred:

```
/.../repos/os/src/drivers/acpi/spec/x86/../../acpi.cc: In static member function 'static void Element::clean_list(Genode::Allocator&)':
/.../repos/os/src/drivers/acpi/spec/x86/../../acpi.cc:1241:74: error: the address of 'Element::_name' will never be NULL [-Werror=address]
 1241 |                                 freed_up += sizeof(*element) + (element->_name ? element->_name_len : 0);
      |                                                                 ~~~~~~~~~^~~~~
/.../repos/os/src/drivers/acpi/spec/x86/../../acpi.cc:741:32: note: 'Element::_name' declared here
  741 |                 char           _name[64];           /* name of element */
      |                                ^~~~~
```

I'm not sure what to do here.

chelmuth, 2023-05-03:
> The check is superfluous and can be completely removed.

Like follows?

```diff
+++ b/repos/os/src/drivers/acpi/acpi.cc
@@ -1238,7 +1238,7 @@ class Element : private List<Element>::Element
                                        continue;
                                }
 
-                               freed_up += sizeof(*element) + element->_name ? element->_name_len : 0;
+                               freed_up += sizeof(*element);
 
                                Element * next = element->next();
                                Element::list()->remove(element);
```

alex-ab, 2023-05-03:
> Like follows?
>
> ```diff
> +++ b/repos/os/src/drivers/acpi/acpi.cc
> @@ -1238,7 +1238,7 @@ class Element : private List<Element>::Element [...]

Yes

commit b09347ca (2023-05-03): acpi_drv: count only memory allocated by new()

Fixes #4844
commit d9643b61 (2023-05-30): acpi_drv: count only memory allocated by new()

Fixes #4844
closed 2023-05-30 by chelmuth