genodelabs/genode #4846: virtualbox6: build errors with gcc 12

cproc, 2023-05-03:
When building `virtualbox6` with gcc 12, the following errors occur:

```
In file included from /.../contrib/virtualbox6-7206beec5f83e36d8ed850537ae730d862e790d5/src/virtualbox6/src/VBox/VMM/include/DBGFInternal.h:31,
                 from /.../contrib/virtualbox6-7206beec5f83e36d8ed850537ae730d862e790d5/src/virtualbox6/src/VBox/VMM/VMMR3/DBGFReg.cpp:24:
In function 'const char* RTStrEnd(const char*, size_t)',
    inlined from 'int dbgfR3RegRegisterCommon(PUVM, PCDBGFREGDESC, DBGFREGSETTYPE, void*, const char*, uint32_t)' at /.../contrib/virtualbox6-7206beec5f83e36d8ed850537ae730d862e790d5/src/virtualbox6/src/VBox/VMM/VMMR3/DBGFReg.cpp:282:44:
/.../contrib/virtualbox6-7206beec5f83e36d8ed850537ae730d862e790d5/src/virtualbox6/include/iprt/string.h:2646:50: error: 'void* memchr(const void*, int, size_t)' specified bound 9223372036854775792 exceeds source size 15 [-Werror=stringop-overread]
 2646 |         char const *pszRet = (char const *)memchr(pszString, '\0', RTSTR_MEMCHR_MAX);
      |                                            ~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```
```
In file included from /.../contrib/virtualbox6-7206beec5f83e36d8ed850537ae730d862e790d5/src/virtualbox6/src/VBox/Runtime/common/string/RTStrSplit.cpp:32:
In function 'const char* RTStrEnd(const char*, size_t)',
    inlined from 'int RTStrSplit(const char*, size_t, const char*, char***, size_t*)' at /.../contrib/virtualbox6-7206beec5f83e36d8ed850537ae730d862e790d5/src/virtualbox6/src/VBox/Runtime/common/string/RTStrSplit.cpp:50:35:
/.../contrib/virtualbox6-7206beec5f83e36d8ed850537ae730d862e790d5/src/virtualbox6/include/iprt/string.h:2646:50: error: 'void* memchr(const void*, int, size_t)' specified bound 9223372036854775792 exceeds source size 15 [-Werror=stringop-overread]
 2646 |         char const *pszRet = (char const *)memchr(pszString, '\0', RTSTR_MEMCHR_MAX);
      |                                            ~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

This error has been fixed upstream with commit https://github.com/mdaniel/virtualbox-org-svn-vbox-trunk/commit/f1f57200a7f5c2c294723030a90defb93c3c05e4

```
/.../contrib/virtualbox6-7206beec5f83e36d8ed850537ae730d862e790d5/src/virtualbox6/src/VBox/Main/src-server/MediumImpl.cpp:70:35: error: 'template<class _Arg, class _Result> struct std::unary_function' is deprecated [-Werror=deprecated-declarations]
   70 |     struct EqualsTo : public std::unary_function <SnapshotRef, bool>
      |                                   ^~~~~~~~~~~~~~
In file included from /.../contrib/stdcxx-b657198052b46799dee7b9c80e910f1d46099e82/include/stdcxx/bits/stl_tree.h:65,
                 from /.../contrib/stdcxx-b657198052b46799dee7b9c80e910f1d46099e82/include/stdcxx/std/map:60,
                 from /.../contrib/virtualbox6-7206beec5f83e36d8ed850537ae730d862e790d5/src/virtualbox6/src/VBox/Main/include/VirtualBoxBase.h:28,
                 from /.../contrib/virtualbox6-7206beec5f83e36d8ed850537ae730d862e790d5/src/virtualbox6/VBoxAPIWrap/MediumWrap.h:31,
                 from /.../contrib/virtualbox6-7206beec5f83e36d8ed850537ae730d862e790d5/src/virtualbox6/src/VBox/Main/include/MediumImpl.h:25,
                 from /.../contrib/virtualbox6-7206beec5f83e36d8ed850537ae730d862e790d5/src/virtualbox6/src/VBox/Main/src-server/MediumImpl.cpp:19:
/.../contrib/stdcxx-b657198052b46799dee7b9c80e910f1d46099e82/include/stdcxx/bits/stl_function.h:117:12: note: declared here
  117 |     struct unary_function
      |            ^~~~~~~~~~~~~~
/.../contrib/virtualbox6-7206beec5f83e36d8ed850537ae730d862e790d5/src/virtualbox6/src/VBox/Main/src-server/MediumImpl.cpp:101:35: error: 'template<class _Arg, class _Result> struct std::unary_function' is deprecated [-Werror=deprecated-declarations]
  101 |     struct EqualsTo : public std::unary_function <BackRef, bool>
      |                                   ^~~~~~~~~~~~~~
/.../contrib/stdcxx-b657198052b46799dee7b9c80e910f1d46099e82/include/stdcxx/bits/stl_function.h:117:12: note: declared here
  117 |     struct unary_function
      |            ^~~~~~~~~~~~~~
```

This error has been fixed upstream with commit https://github.com/mdaniel/virtualbox-org-svn-vbox-trunk/commit/6060cfff7310c641ca254d53501988b681552bee, but when I applied it, the `#if` condition did not match. It only disables the error message though, which can also be done via `CC_WARN`.

```
/.../contrib/virtualbox6-7206beec5f83e36d8ed850537ae730d862e790d5/src/virtualbox6/src/VBox/Devices/PC/DevFwC
ommon.cpp: In function 'int FwCommonPlantDMITable(PPDMDEVINS, uint8_t*, unsigned int, PCRTUUID, PCFGMNODE, uint16_t, uint16_t*, uint16_t*, bool)':
/.../contrib/virtualbox6-7206beec5f83e36d8ed850537ae730d862e790d5/src/virtualbox6/src/VBox/Devices/PC/DevFwCommon.cpp:534:38: error: writing 1 byte into a region of size 0 [-Werror=stringop-overflow=]
  534 |         *pszStr++                    = '\0'; /* terminate set of text strings */ \
      |         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~
/.../contrib/virtualbox6-7206beec5f83e36d8ed850537ae730d862e790d5/src/virtualbox6/src/VBox/Devices/PC/DevFwCommon.cpp:820:9: note: in expansion of macro 'DMI_TERM_STRUCT'
  820 |         DMI_TERM_STRUCT;
      |         ^~~~~~~~~~~~~~~
/.../contrib/virtualbox6-7206beec5f83e36d8ed850537ae730d862e790d5/src/virtualbox6/src/VBox/Devices/PC/DevFwCommon.cpp:197:21: note: destination object 'DMICHASSIS::u32OEMdefined' of size 4
  197 |     uint32_t        u32OEMdefined;
      |                     ^~~~~~~~~~~~~
```

This error has been fixed upstream with commit https://github.com/mdaniel/virtualbox-org-svn-vbox-trunk/commit/446b5d5a350eb8cc010c3128a60ee7644c661e4a, but this fix uses the `mempcpy()` function, which is currently not available on Genode.

chelmuth, 2023-05-03:
@cproc are the mentioned commits part of 6.1.40 (or later 6.1) as in issue #4666?

cproc, 2023-05-03:
> @cproc are the mentioned commits part of 6.1.40 (or later 6.1) as in issue #4666?

The RTStrEnd commit is part of 7.0.0 and later and the other two commits are part of 6.1.42 and later.

closed 2023-05-30 by chelmuth