genodelabs/genode #4851: sd_card_drv: build errors with gcc 12

cproc, 2023-05-03:
When building `drivers/sd_card`, the following error occurs:

```
In file included from /.../repos/os/src/drivers/sd_card/imx53/driver.cc:15:
/.../repos/os/src/drivers/sd_card/imx/driver.h:201:59: error: reference to 'Mmio' is ambiguous
  201 |                 struct Timer_delayer : Timer::Connection, Mmio::Delayer
      |                                                           ^~~~
In file included from /.../repos/os/include/platform_session/device.h:18,
                 from /.../repos/os/src/drivers/sd_card/imx/driver.h:18:
/.../repos/base/include/util/mmio.h:77:1: note: candidates are: 'struct Genode::Mmio Genode::Mmio::Mmio'
   77 | {
      | ^
/.../repos/os/include/platform_session/device.h:90:1: note:                 'class Platform::Device::Mmio Platform::Device::Mmio::Mmio'
   90 | {
      | ^
/.../repos/os/include/platform_session/device.h:89:25: note:                 'class Platform::Device::Mmio'
   89 | class Platform::Device::Mmio : Range, Attached_dataspace, public Genode::Mmio
      |                         ^~~~
```

and more errors of the same type.

cproc, 2023-05-03:
Fixed by b57b674.

nfeske, 2023-05-03:
I find the spilling of the `Genode::` prefixes unfortunate.

If so many places must be prefixed, could the disambiguation be solved via a `using` declaration in an appropriate scope where `Mmio` is used? Like this:

```
using Mmio = Genode::Mmio;
```

nfeske, 2023-05-03:
Thanks for giving it a try. That's good to read.

Maybe this way can be considered in other patches where several prefixes were added?

It would be nice to keep the changes minimal and generally avoid noisy namespace prefixing as a matter of coding style.

commit 030b879b (2023-05-03): sd_card_drv: fix build errors with gcc 12

Fixes #4851
commit 901cbf0a (2023-05-30): sd_card_drv: fix build errors with gcc 12

Fixes #4851
closed 2023-05-30 by chelmuth