genodelabs/genode #4857: dde_bsd: OpenBSD 7.1 no longer available for download

chelmuth, 2023-05-04:
The official server (https://cdn.openbsd.org/pub/OpenBSD/7.1/sys.tar.gz) no longer provides OpenBSD 7.1 for download. As long as some mirrors still provide the archive (e.g., https://ftp.fau.de/pub/OpenBSD/7.1/sys.tar.gz), we could just change the URL. The other option is an update to 7.2 or 7.3.

@cnuke what's your stance?

cnuke, 2023-05-04:
@chelmuth I would squeeze in updating the driver as the diff is small and the update brings in support (mostly adding PCI IDs) for newer platforms.

cnuke, 2023-05-04:
Commit [a61fdf7](https://github.com/cnuke/genode/commit/a61fdf72c169f61e275ccb8f503efd002d0ac370) updates the driver to version 7.3 (and thereby fixes this issue).

I have briefly tested (`audio_out.run` and `audio_in.run`) the new version on the X201, Fuji4 and Fuji6 where it worked as advertised.

chelmuth, 2023-05-04:
> I have briefly tested (`audio_out.run` and `audio_in.run`) the new version on the X201, Fuji4 and Fuji6 where it worked as advertised.

Thanks. Audio-In on U7412 worked?

cnuke, 2023-05-04:
@chelmuth no, the recording is still unsupported. Sorry for not being explicit about that.

chelmuth, 2023-05-04:
Nevermind, that's what I expected anyway ;-)

commit a61fdf72 (2023-05-04): dde_bsd: update audio driver to OpenBSD 7.3

Fixes #4857.
commit 03ed5fc3 (2023-05-30): dde_bsd: update audio driver to OpenBSD 7.3

Fixes #4857.
closed 2023-05-30 by chelmuth