genodelabs/genode #4858: Sculpt: prematurely entering ENTER might get the network dialog stuck

cnuke, 2023-05-04:
While using the PinePhone I noticed that if I input less than the require number of characters (8) and press enter, which triggers a connect attempt, the dialog might get stuck displaying *connecting*. When this happens I am not able to trigger the AP selection again.

cnuke, 2023-05-04:
Commit [23d7e20](https://github.com/cnuke/genode/commit/23d7e20f27e8d667ffa48fc59d98d5980558ff60) provides a quick-fix.

(I will look into if we could make the driver provde more feedback in such situations to better assist a management component.)

nfeske, 2023-05-04:
Thank you for the report and the immediate fix. I merged your commit to staging and to my [sculpt_23_04](https://github.com/nfeske/genode/commits/sculpt_23_04) and [sculpt_23_04_pinephone](https://github.com/nfeske/genode/commits/sculpt_23_04_pinephone) branches (that I use for backporting selected patches from mainline to Sculpt 23.04).

nfeske, 2023-05-30:
Fixed in master.

closed 2023-05-30 by nfeske