genodelabs/genode #4861: Move wifi-stack into the dde_linux repository

cnuke, 2023-05-08:
For historical reasons the wifi-stack was tailored to the PC platform and up to now was located in the _pc_ repository. To ease supportting other platforms and other transport layers besides PCIe the common portions are going to be moved into the _dde_linux_ repository while keeping the platform-specific parts in the corresponding platform-repository

cnuke, 2023-05-11:
The following commits contain

* move from the _pc_ to _dde_linux_ repository
* firmware loading via the VFS (from a locally attached `.tar` archive acquired as ROM module as an example)
* `wpa_supplicant` update to 2.10 (I briefly attemptted a `libnl` update as well but decided against it for now as it appears to rely on more Linux infrastructure and the new supplicant still works with the old one)

and the major changes are documented in the commit messages as well in the various _README_ files. Commits logically contained by another one are denoted as _fixup_ commit to make the changes prominent.

The `pc_wifi` as well as the `a64_wifi` driver pkgs, the various run-scripts and the Sculpt integration were adapted. Initial testing was performed on ultra, macho, fuji4, fuji6 and x201 (_pc_wifi.run_, _netperf_lwip_wifi_, Sculpt (fuji* only)) as well as on the PinePhone (_a64_wifi.run_, Sculpt) and did not exhibit many unexpected (*) results.

(*) ultra still triggers a page-fault (I am going to further look into that for the release) and fuji6 sometimes mis-behaves when connected to our 2.4GHz test network (frequent disconnects when connected)

I plan to further refine the `wifi_drv` for the upcoming release but these are mostly internal adjustments and no functional change noticable from the out-side and should be on the “low-risk” side.

Genode repository:

* 2fe8bbd815 - dde_linux: remove unused var from libnl_include
* f23280cd24 - dde_linux: update wpa_supplicant to version 2.10
* 6fd31df773 - fixup 'wifi: get firmware from to tar archive' (Sculpt adaptions)
* 99437ff172 - wifi: get firmware from to tar archive
* 3446b56e87 - fixup 'Move 'wifi' driver to dde_linux repository' (Sculpt adaptions)
* aee785f10b - Move 'wifi' driver to dde_linux repository
* 3b976033ff - dde_linux: promote PC specific additions

Allwinner repository:
* 2bc5828 - Adapt to wpa_supplicant update
* efb4144 - Adapt to firmware from tar archive
* 6a2d2f5 - phone_manager: adapt to renaming
* 70166ba - a64_wifi_drv: adapt to wifi move
* d70cbce - a64_wifi_drv: remove wrong api/pc_linux entry
* c3c4993 - Adapt to promote PC specific additions

cnuke, 2023-05-22:
The following commit series provides the changes.

Genode repository:
* b585cd0bc9 - dde_linux: promote PC specific additions
* d2ee9b3b47 - Move 'wifi' driver to dde_linux repository
* 9374a8743b - fixup 'Move 'wifi' driver to dde_linux repository' (Sculpt adaptions)
* d629d1261b - pc_wifi: implement __crypto_xor generated dummy
* c2f89537e2 - wifi: get firmware from to tar archive
* 07a21ef6a5 - fixup 'wifi: get firmware from to tar archive' (Sculpt adaptions)
* 31a1212e3c - dde_linux: update wpa_supplicant to version 2.10
* 7e354c8b5b - dde_linux: remove unused var from libnl_include
* aece9568a6 - wifi: add RFKILL notification interface
* 70199ff7ee - pci_decode: force GSI for ath9k device
* e11016e117 - dde_linux: enhance KBUILD_MODNAME generation
* 9cd313c29b - pc_wifi: enable RTL8192CE support in rtlwifi
* 04349de53a - wifi: provide WPA3-personal support 

Allwinner repository:
* c8dcc2a - Adapt to promote PC specific additions
* a4a22c4 - a64_wifi_drv: remove wrong api/pc_linux entry
* f6277dc - a64_wifi_drv: adapt to wifi move
* 268fbc4 - phone_manager: adapt to renaming
* ab0a7fc - Adapt to firmware from tar archive
* 91d03f8 - Adapt to wpa_supplicant update

 @nfeske commit 9374a8743b and 07a21ef6a5 are the ones I was referring to.

chelmuth, 2023-05-23:
@skalk do you approve the dde_linux adaptions in the commit series? I plan to merge it today.

chelmuth, 2023-05-23:
@cnuke do the adaptions in genodelabs/genode affect genode-imx etc.?

cnuke, 2023-05-23:
@chelmuth maybe - I have not checked that as @skalk probably already made the adaptions. But _zynq_ could be a problem, I am going to check that.

skalk, 2023-05-23:
@chelmuth @cnuke I very much like the commit series apart from the minor one-liner comment I made. Just for the record: at least the genode-imx and genode-zynq repositories have to be searched through for further occurrences of the functions now moved to `lx_emul/shadow`, like @cnuke already did for the genode-allwinner repo.

cnuke, 2023-05-23:
> @chelmuth @cnuke I very much like the commit series apart from the minor one-liner comment I made. Just for the record: at least the genode-imx and genode-zynq repositories have to be searched [...]

Right, I'll look into the _imx_ repo as well.

chelmuth, 2023-05-24:
The *builder* produces linker errors on arm_v6/7a.

cnuke, 2023-05-25:
> 

I'll look into that.

cnuke, 2023-05-25:
@chelmuth please merge [5a7cb04](https://github.com/cnuke/genode/commit/5a7cb04bbd05cb29e3be8a6d463db06bafce85ad), [ab4956c](https://github.com/cnuke/genode/commit/ab4956c025fcee876d95e29e83eade17fa12375a) and [95d80e5](https://github.com/cnuke/genode/commit/95d80e57738b8203c91d72bc82e15a8ae3b19313).

(I have noticed several `<policy>` relating warnings in the _nic_router_uplinks.run_ run-script that I ignored as it appears to work as intended.)

nfeske, 2023-05-30:
Fixed in master.

closed 2023-05-30 by nfeske