genodelabs/genode #4862: libports: Memory leak in mesa (21.0.0) on Broadwell platforms

ssumpf, 2023-05-08:
When the HUD is enabled to display, for example, the current FPS in the `<config>` node of an 3D app

`<env key="GALLIUM_HUD" value="fps"/>`

it leads to a memory leak of 64 Byte per frame on GEN8 GPUs (Broadwell) only. Since the HUD is meant for debugging/analyzing, one fix is to remove the `GALLIUM_HUD` from the application. The leak is caused in the function `iris_create_surface` (src/gallium/drivers/iris/iris_state.c of our Mesa 21.0.0 port) which calls

```
#if GEN_GEN == 8
alloc_surface_states(...
```` 
which is not freed later on.

nfeske, 2024-05-30:
@ssumpf is this issue still relevant after the update to Mesa 24?

ssumpf, 2024-05-30:
@nfeske: From the source code level it still seems to be an issue. However, I can test it in the office next time and fix it there.

nfeske, 2025-04-29:
@ssumpf is this issue still present?

nfeske, 2025-11-28:
@ssumpf do you foresee any further activity regarding this issue? If not so, I'd suggest to close it.