genodelabs/genode #4864: sel4: build errors with gcc 12

cproc, 2023-05-09:
When building `KERNEL=sel4 BOARD=pc lib/kernel-sel4-pc` with gcc 12, the following errors occur:

```
 [Circular includes] kernel_all.c_pp
Traceback (most recent call last):
  File "/.../contrib/sel4-9ab19376285d865052c2b5021560306306bf19a8/src/kernel/sel4/tools/circular_includes.py", line 70, in <module>
    sys.exit(main())
  File "/.../contrib/sel4-9ab19376285d865052c2b5021560306306bf19a8/src/kernel/sel4/tools/circular_includes.py", line 58, in main
    while file_stack[-1] != header:
IndexError: list index out of range
```

```
/.../contrib/sel4-9ab19376285d865052c2b5021560306306bf19a8/src/kernel/sel4/src/arch/x86/kernel/cmdline.c: In function 'cmdline_parse':
/.../contrib/sel4-9ab19376285d865052c2b5021560306306bf19a8/src/kernel/sel4/src/arch/x86/kernel/cmdline.c:117:34: error: array subscript 0 is outside array bounds of 'const short unsigned int[0]' [-Werror=array-bounds]
  117 |     int const bda_ports_count       = (*bda_equi >> 9) & 0x7;
      |                                  ^~~~~~~~~
/.../contrib/sel4-9ab19376285d865052c2b5021560306306bf19a8/src/kernel/sel4/src/arch/x86/kernel/cmdline.c:122:52: error: array subscript 0 is outside array bounds of 'const short unsigned int[0]' [-Werror=array-bounds]
  122 |     cmdline_opt->console_port = bda_ports_count && *bda_port ? *bda_port : 0x3f8;
      |                                                    ^~~~~~~~~
/.../contrib/sel4-9ab19376285d865052c2b5021560306306bf19a8/src/kernel/sel4/src/arch/x86/kernel/cmdline.c:144:50: error: array subscript 0 is outside array bounds of 'const short unsigned int[0]' [-Werror=array-bounds]
  144 |     cmdline_opt->debug_port = bda_ports_count && *bda_port ? *bda_port : 0x3f8;
      |                                                  ^~~~~~~~~
```

cproc, 2023-05-09:
Fixed by ac5cd2f (backported upstream fixes).

chelmuth, 2023-05-09:
@cproc @alex-ab thanks, f311d366d52bd7568eb9231fc098f1397f0c04a6 fixes sel4 kernel build for me.

commit f311d366 (2023-05-09): sel4: fix build errors with gcc 12

Fixes #4864
closed 2023-05-30 by chelmuth