genodelabs/genode #4866: depot/remove

a-dmg, 2023-05-09:
Dear Genodians,

Following [our discussion](https://lists.genode.org/pipermail/users/2023-March/008529.html) on the mailing list. I propose implementing the 'depot_remove'  component to remove PKGs from the depots and resolve dependencies that are not required anymore and that can be deleted.

Here are the thought of Norman's about such component:
```
Let me share my thoughts what I would expect from a depot-uninstall 
component. In general, it would perform two steps.

1. It would remove a selection of depot packages according to its
    configuration. With depot package, I only refer to pkg/<version>/
    directories.

2. It would garbage-collect all depot content that is no longer
    referenced by any pkg present in the depot. I guess this is what
    you had in mind with the 'depot_autoremove' naming.

The pkg-removal step raises the question of how to specify the set of 
packages to remove. You suggested specifying a list of pkgs to keep and 
discard everything not featured in this list. In other situations, like 
when using Sculpt with a large disk and preserving the ability to roll 
the system back to any previous version, it would be more appropriate to 
explicitly select the packages to remove - letting the user take 
interactive decisions.

I think the <config> of the depot-uninstall tool could accommodate both 
situations quite well. E.g.,

   Remove one specific version of a pkg:

   <config>
     <remove user="cnuke" pkg="pdf_view" version="2022-02-22"/>
     ...
   </config>

   Remove all versions of a pkg:

   <config>
     <remove user="cnuke" pkg="pdf_view"/>
     ...
   </config>

   Remove all pkgs of a specified depot user:

   <config>
     <remove user="cnuke"/>
     ...
   </config>

   Remove all pkgs except for an explicit selection of packages to keep:

   <config>
     <remove-all>
       <keep user="cnuke" pkg="pdf_view"/>
     </remove-all>
     ...
   </config>

   The <keep> node could also give the freedom to select a particular
   version or a whole user.

```

Cheers,
Alice

nfeske, 2023-06-08:
Thanks Alice for implementing the ideas and proposing the results of your work as such a complete package (along with documentation and even a depot recipe). I'm happy the see that the feature can be implemented at a rather low complexity, which is what I hoped for. I also appreciate the effort you have put into the automated test, which nicely serves as an additional point of documentation. The multi-stage testing by spawning variants of the same scenario is nifty!

Most importantly, the run script works beautifully. :-)

With commit b35286dcad0e3b3175bd9f3a52a3ca179eab414e, I just added a small fixup for the README.

Towards merging the commit, I'd like to encourage you to consider the following points. Some of them are fairly subjective. I hope you don't take my review as overly picky. It's just me trying to establish a consistent coding style throughout our code base. Should you not want to apply these changes, I'd be also up to it but only if get your green light. Please let me know how you prefer to proceed.

* The indentation of the block comments is not consistent with the surrounding code. But maybe, the block comments could be dropped anyway? At least to me during the review, they played no important role.

* There is no need to mark methods as 'inline' whenever the implementation is directly provided with the declaration.

* Private methods should be prefixed with `_`. I tend to also use the `_` prefix for members that are intended as being private, as oposed to being a public interface. E.g., `Main` has no public interface, so I usually prefix its members with `_`, even when `Main` is technically a `struct`.

* The `Genode::` prefix can be removed in some places.

* Would it be possible to use a `Registry` instead of a `List`? The `Registry` is safer because no pointers must be handled manually and the element lifetimes remain always consistent with the data structure. This change would also alleviate the need for `for_each_path_list_elem`.

* The `Archive_remover::_arch` member can be const.

* It's good practice to put control-flow-altering statements (`return`, `break`, `continue`) at a new line to make it easier to follow the flow.

* Inside the `Archive_remover`, you may replace `Heap` by `Allocator` because the `Archive_remover` requires only the `Allocator` interface.

* I recommend removing the `_reporter` member from `Archive_remover`. Instead, you may pass an `Xml_generator &` as argument to the `generate_report` method.  This way, the distinction whether to generate the report or not can happen at the caller site by calling the method or not. The method may then even become `const`, which clarifies that the generation of the report does not change any state (it only reads).  You may also reconsider the other reference members, in particular the `Directory`. By passing references explicitly to only the methods that need them, the possible side effects of each method become more obvious. I'd probably keep the `Allocator` as reference member to ensure that `new` and `destroy` consistently use the same allocator.

* Since the `Archive_remover::generate_report` method is only called by the `Archive_remover`, you may make it private. The same would apply for `configure_remove_all_pkgs` and `configure_remove_pkgs`.

* You may consider replacing some of the `error` calls by `warning` whenever the component can gracefully handle it.

* It would be nice to avoid putting closing braces at newlines in situations where they to not contribute to the reader's sense of scope. E.g., I'd probably put all the closing braces and parenthesis inside `for_each_pkg` at the end of the `fn(pkg_version_path)` line. It reads just as well and is more succinct. To improve readability further, you may consider renaming `pkg_dir` to `pkg` and `depot_user_pkg_path` to `pkg_dir` (no need to encode the type in the variable name). By `using Path = Directory::Path`, you can further avoid the repetitive use of the long name, making the interesting parts of the code stand out more. With these cosmetic changes combined, the code should become almost as nice to read as prose.

* Can the deeply nested condition inside `configure_remove_all_pkgs` be simplified by putting it into a local lamba, and negating some of the conditions? This would make the logic easier digestible. To illustrate the idea:

  ```
  auto keep = [&]
  {
    if (!node.has_attribute("user"))
      return false;
  
    if (...)
      return false;
  
    return true;
  };
  
  if (!keep())
    ...
  ```

* The inversion of conditions (and early returns) may also help in other places. E.g., in `configure_remove_pkgs`, by immediatly returning whenever no user attribute is present, the indentation of all following lines can be be reduced.

* I wondered, since the lifespan of `archive_cleaner` does not exceed the scope of `handle_config`, couldn't it become a local variable.

nfeske, 2023-06-14:
Thank you for following my recommendations. It's very good to see the gain in simplicity.

> But I still need to change the closing braces at Newline as I am not sure to understand.

It is actually a fairly mundane point. I would format the following method
```
template<typename FUNC = void(Path const &)>
void _for_each_pkg(Directory &depot, FUNC fn)
{
    depot.for_each_entry([&](auto const &entry) {
        Path pkg_path { entry.name(), "/pkg" };
        if (depot.directory_exists(pkg_path)) {
            _for_each_subdir(depot, pkg_path, [&] (Path const &pkg_path) {
                _for_each_subdir(depot, pkg_path, [&] (Path const &pkg_version_path) {
                    fn(pkg_version_path);
                });
            });
        }
    });
}
```
to the following form:
```
template<typename FUNC = void(Path const &)>
void _for_each_pkg(Directory &depot, FUNC fn)
{
    depot.for_each_entry([&](auto const &entry) {
        Path pkg_path { entry.name(), "/pkg" };
        if (depot.directory_exists(pkg_path)) {
            _for_each_subdir(depot, pkg_path, [&] (Path const &pkg_path) {
                _for_each_subdir(depot, pkg_path, [&] (Path const &pkg_version_path) {
                    fn(pkg_version_path); }); }); } });
}
```
because the scopes are so obvious.

Thank you for following my recommendations. It's very good to see the gain in simplicity.

Let me propose the following few finishing touches:

* Add the following line to the `Archive_remover`:
  ```
  using Registered_path = Registered_no_delete<Archive_path>;
  ```
  and replace all occurrences of the of complicated looking `Registered_no_delete<Archive_path>` by the `Registered_path`. The resulting code becomes much easier on the eyes.

* Remove the use of `reinterpret_cast` by using `Registered_path` as lambda argument, like this:
  ```
  _archive_to_delete.for_each([&](Registered_path &path){
      if (dependency_path == path)
          destroy(_alloc, &path); });
  ````

* Put opening braces of methods consistently at the beginning of a new line.

* Review horizontal alignment (e.g., looking at the `Archive_remover` constructor arguments)

* For the conditional construction of the reporter, you may find the `Constructible::conditional` helper useful. It helps to make the code simpler:
  ```
  _reporter.conditional(config.attribute_value("report", false),
                        _env, "removed_archives", "archive_list");

  if (_reporter.constructed())
      archive_cleaner.generate_report(*_reporter);
  ```

* We usually do not capitalize warning messages and don't use punctuation for warning messages and single-line comments.

* It would be nice adding a file-creation date at the file header (just for consistency).

a-dmg, 2023-06-14:
> But I still need to change the closing braces at Newline as I am not sure to understand.

Thank you for clarifying!

I have addressed all your latest remarks into an additional fixup commit.

nfeske, 2023-06-14:
That was quick! Thanks a lot for your patience about my coding-style nitpicks and the fixup. :-)

I merged the result (all commits squashed into one) as https://github.com/genodelabs/genode/commit/305ba164c2b7c8be85b5615bfb28e2884fe03b27 to staging now.

Since you provided such a nice automated test - including the assert-guard for linux - I added the run script to the tool/autopilot.list so that we can keep an eye on it.

nfeske, 2023-06-15:
The depot_remove.run script succeeded this night for Linux on 64-bit. However, 32-bit Linux, produces the following output:

```
spawn ./core
Genode 23.05-80-g566d1dfe3b4[0m
4095 MiB RAM and 18997 caps assigned to init[0m
[init -> lx_fs] [34mWarning: ram donation is 155000 but used RAM is 155648B, 'depot_remove -> '[0m[0m
[init -> report_rom] report 'depot_remove -> archive_list'[0m[0m
[init -> report_rom]   <removed_archives>[0m[0m
[init -> report_rom]   	<removed path="checker/pkg/nano3d/2023-06-14"/>[0m[0m
[init -> report_rom]   </removed_archives>[0m TEST 1 --- Delete nano3d -- SUCCESS
spawn ./core
Genode 23.05-80-g566d1dfe3b4[0m
4095 MiB RAM and 18997 caps assigned to init[0m
[init -> lx_fs] [34mWarning: ram donation is 155000 but used RAM is 155648B, 'depot_remove -> '[0m[0m
[init -> report_rom] report 'depot_remove -> archive_list'[0m[0m
[init -> report_rom]   <removed_archives/>[0m[0m
ERROR: checker/bin/x86_64/backdrop/2023-06-14 should still be there but it has been deleted.
 TEST 2 --- Delete non existing archive -- ERROR
make[1]: *** [Makefile:431: run/depot_remove] Error 1
make[1]: Leaving directory '/data/autopilot/autopilot.checker.2023-06-15-00-12/linux/x86_32'
```

nfeske, 2023-06-16:
The problem seems to be that the `create_tar_from_depot_binaries` function populates the .tar archive with binaries for the architecture matching the build directory whereas the run script has `x86_64` hard-coded at several places. So the test should be expected to fail also on e.g., Linux on arm_v8a.

The cleanest solution would be replacing the hard-coded arch values by calls of the `depot_spec` function (defined at tool/run/depot.inc).

nfeske, 2023-06-16:
I've added fixup https://github.com/genodelabs/genode/commit/e120107b9ba3eb5685228ea65f0ec8b20c6f4393 as an intermediate step, disabling the test on all architectures other than x86_64.

I'd very much appreciate a follow-up commit making the test CPU-architecture-agnostic.

nfeske, 2023-06-16:
The feature entered the master branch. Let's keep the issue open until the `depot_spec` adaptation is wrapped up.

a-dmg, 2023-06-19:
I have provided a fixup commit that addresses uses of `[depot_spec]` instead of hardcoded spec. This should fix the issue on `x86_32-linux`.

nfeske, 2023-06-20:
Thank you for the adjustments, @a-dmg.

Since the original commit already entered the master branch, I gave your patch a new commit message and merged it as https://github.com/genodelabs/genode/commit/98b32aa042e203ba6c1dfcb1b3d2dab6b900d217 to staging. Do you approve?

While giving it a quick spin, I noticed the output at the very end:

```
ERROR: nfeske_test/bin/x86_64/themed_decorator/2023-06-20 is present but should has been deleted.
 TEST 4 --- Remove all, keep themed_decorator PKG --- SUCCESS
```

I just wondered, is the "ERROR" expected?

a-dmg, 2023-06-20:
Thanks, @nfeske. Yes, I approve!  

Yes, the error from the `check_depot_state` function is expected.  

1. The component must delete everything but the `themed_decorator` PKG.
2. We want to test that even if everything is requested to be deleted but one PKG, we verify that this PKG and its dependency are still kept.  
3. To avoid duplicating too much code in the runcript, I reused the `depot_state_for_pkg` function that registers the state of a PKG and its dependencies for deletion.  Thus `check_depot_state` expect that this PKG and its dependencies are deleted. But they are not if the functionality works properly.
4. The test succeeds if `check_depot_state` returns an error, displaying the error message you notice.

While this is convenient for a lazy person like me, it isn't for the readability of the test and ease of understanding what's going on. I see serval points of improvement for the runscript:  

 - Instead of duplicating code between `check_depot_state`, `depot_state_for_pkg` and `depot_state`, only one function could be written to traverse the depot. And other functions only manipulate the `$context` dictionary to register the package in the correct sublist properly.
 - Function can be renamed to reflect better what they do, such as `register_for_deletion`, `register_for_conservation`...
 - Brief comments can be added to describe what functions do and how tests work
 - Test 4 does test that a specific PKG is kept when everything else is deleted. But it does not verify that everything else has been deleted. So it can be extended, or an additional test can be added.
 - Typos could be corrected
 
I would provide an extra commit to address this. Do you agree? Or would you like to have one commit, including the depot_spec fix?

a-dmg, 2023-06-23:
Thank you, @nfeske, for the precision about the commit messages!  
  
 I have pushed two extra commits ([e47e881](https://github.com/a-dmg/genode/commit/e47e881aa8cc25994f64f85e5b870ccc03536e15) and [1986675](https://github.com/a-dmg/genode/commit/19866753d349ef991d2eb88760a7d5170171925b)) that address the points I raised with my last comment.

nfeske, 2023-06-28:
Your refinements look good. However, before merging the commits, I gave the new version a quick try on x86_64 (tried on staging and master), which produces the following error now. Can you reproduce it? Or should I provide you with the full log?

```
--------------- TEST 3 --- Delete a PKG archive with deps to keep
spawn ./core
Genode 23.05-58-gb49ab9fb3bf
17592186044415 MiB RAM and 18997 caps assigned to init
[init -> report_rom] report 'depot_remove -> archive_list'
[init -> report_rom]   <removed_archives>
[init -> report_rom]   	<removed path="nfeske_test/raw/ttf-bitstream-vera-minimal/2019-02-25"/>
[init -> report_rom]   	<removed path="nfeske_test/raw/fonts_fs/2019-02-25"/>
[init -> report_rom]   	<removed path="nfeske_test/bin/x86_64/vfs_ttf/2023-06-15"/>
[init -> report_rom]   	<removed path="nfeske_test/pkg/fonts_fs/2023-06-15"/>
[init -> report_rom]   </removed_archives>
ERROR: nfeske_test/raw/fonts_fs/2019-02-25 should still be there but it has been deleted.
 TEST 3 --- Delete a PKG archive with deps to keep --- ERROR
make: *** [Makefile:431: run/depot_remove] Error 1
```

As rather cosmetic detail, would you consider fixup b49ab9fb3bfedd1557754671b3a4140640e47570? The quoting of the glob argument as string allows the syntax highlighting of vim to work better.

a-dmg, 2023-07-06:
I have a hard time trying to reproduce this issue. However, Pirmin has confirmed that he can produce a similar error on that test (but a different PKG). Could you please share more information on your build configuration?  

We have yet to be able to understand why this seems machine dependent. We both tried on our current dev environment and a fresh genode clone.

nfeske, 2023-07-12:
I have added a bit of instrumentation with commit https://github.com/nfeske/genode/commit/1660e96a8b6fb49b0d7f0a7634bf791d8f8381f9, printing the depot content between the tests. Here is the output on my machine:

```
--------------- TEST 1 --- Delete nano3d
current depot content:
var/run/depot_remove/genode/depot/nfeske_test/raw/sticks_blue_backdrop/2019-02-25
var/run/depot_remove/genode/depot/nfeske_test/raw/genode_bg/2019-02-25
var/run/depot_remove/genode/depot/nfeske_test/raw/wm/2020-06-21
var/run/depot_remove/genode/depot/nfeske_test/raw/window_layouter/2020-02-19
var/run/depot_remove/genode/depot/nfeske_test/raw/ttf-bitstream-vera-minimal/2019-02-25
var/run/depot_remove/genode/depot/nfeske_test/raw/fonts_fs/2019-02-25
var/run/depot_remove/genode/depot/nfeske_test/raw/system_shell/2021-10-14
var/run/depot_remove/genode/depot/nfeske_test/bin/x86_64/vfs_pipe/2023-06-15
var/run/depot_remove/genode/depot/nfeske_test/bin/x86_64/libpng/2023-05-26
var/run/depot_remove/genode/depot/nfeske_test/bin/x86_64/nano3d/2023-06-15
var/run/depot_remove/genode/depot/nfeske_test/bin/x86_64/themed_decorator/2023-06-15
var/run/depot_remove/genode/depot/nfeske_test/bin/x86_64/vfs_ttf/2023-06-15
var/run/depot_remove/genode/depot/nfeske_test/bin/x86_64/report_rom/2023-06-15
var/run/depot_remove/genode/depot/nfeske_test/bin/x86_64/vfs/2023-06-15
var/run/depot_remove/genode/depot/nfeske_test/bin/x86_64/wm/2023-06-15
var/run/depot_remove/genode/depot/nfeske_test/bin/x86_64/window_layouter/2023-06-15
var/run/depot_remove/genode/depot/nfeske_test/bin/x86_64/decorator/2023-06-15
var/run/depot_remove/genode/depot/nfeske_test/bin/x86_64/libc/2023-06-15
var/run/depot_remove/genode/depot/nfeske_test/bin/x86_64/bash-minimal/2023-06-28
var/run/depot_remove/genode/depot/nfeske_test/bin/x86_64/init/2023-06-15
var/run/depot_remove/genode/depot/nfeske_test/bin/x86_64/backdrop/2023-06-15
var/run/depot_remove/genode/depot/nfeske_test/bin/x86_64/chroot/2023-06-15
var/run/depot_remove/genode/depot/nfeske_test/bin/x86_64/terminal/2023-06-15
var/run/depot_remove/genode/depot/nfeske_test/bin/x86_64/fs_report/2023-06-15
var/run/depot_remove/genode/depot/nfeske_test/bin/x86_64/coreutils-minimal/2023-06-28
var/run/depot_remove/genode/depot/nfeske_test/bin/x86_64/vim-minimal/2023-06-28
var/run/depot_remove/genode/depot/nfeske_test/bin/x86_64/zlib/2023-05-26
var/run/depot_remove/genode/depot/nfeske_test/bin/x86_64/fs_rom/2023-06-15
var/run/depot_remove/genode/depot/nfeske_test/bin/x86_64/ncurses/2023-06-28
var/run/depot_remove/genode/depot/nfeske_test/bin/x86_64/posix/2023-06-15
var/run/depot_remove/genode/depot/nfeske_test/bin/x86_64/cached_fs_rom/2023-06-15
var/run/depot_remove/genode/depot/nfeske_test/pkg/sticks_blue_backdrop/2023-06-15
var/run/depot_remove/genode/depot/nfeske_test/pkg/nano3d/2023-06-15
var/run/depot_remove/genode/depot/nfeske_test/pkg/themed_decorator/2023-06-15
var/run/depot_remove/genode/depot/nfeske_test/pkg/wm/2023-06-15
var/run/depot_remove/genode/depot/nfeske_test/pkg/window_layouter/2023-06-15
var/run/depot_remove/genode/depot/nfeske_test/pkg/backdrop/2023-06-15
var/run/depot_remove/genode/depot/nfeske_test/pkg/chroot/2023-06-15
var/run/depot_remove/genode/depot/nfeske_test/pkg/fonts_fs/2023-06-15
var/run/depot_remove/genode/depot/nfeske_test/pkg/motif_decorator/2023-06-15
var/run/depot_remove/genode/depot/nfeske_test/pkg/system_shell/2023-06-28
spawn ./core
Genode 23.05-58-gb49ab9fb3bf[0m
17592186044415 MiB RAM and 18997 caps assigned to init[0m
[init -> report_rom] report 'depot_remove -> archive_list'[0m[0m
[init -> report_rom]   <removed_archives>[0m[0m
[init -> report_rom]   	<removed path="nfeske_test/bin/x86_64/nano3d/2023-06-15"/>[0m[0m
[init -> report_rom]   	<removed path="nfeske_test/pkg/nano3d/2023-06-15"/>[0m[0m
[init -> report_rom]   </removed_archives>[0m[0m
 TEST 1 --- Delete nano3d -- SUCCESS

--------------- TEST 2 --- Delete non existing archive
current depot content:
var/run/depot_remove/genode/depot/nfeske_test/raw/sticks_blue_backdrop/2019-02-25
var/run/depot_remove/genode/depot/nfeske_test/raw/genode_bg/2019-02-25
var/run/depot_remove/genode/depot/nfeske_test/raw/wm/2020-06-21
var/run/depot_remove/genode/depot/nfeske_test/raw/window_layouter/2020-02-19
var/run/depot_remove/genode/depot/nfeske_test/raw/ttf-bitstream-vera-minimal/2019-02-25
var/run/depot_remove/genode/depot/nfeske_test/raw/fonts_fs/2019-02-25
var/run/depot_remove/genode/depot/nfeske_test/raw/system_shell/2021-10-14
var/run/depot_remove/genode/depot/nfeske_test/bin/x86_64/vfs_pipe/2023-06-15
var/run/depot_remove/genode/depot/nfeske_test/bin/x86_64/libpng/2023-05-26
var/run/depot_remove/genode/depot/nfeske_test/bin/x86_64/themed_decorator/2023-06-15
var/run/depot_remove/genode/depot/nfeske_test/bin/x86_64/vfs_ttf/2023-06-15
var/run/depot_remove/genode/depot/nfeske_test/bin/x86_64/report_rom/2023-06-15
[... 101 lines cut ...]
var/run/depot_remove/genode/depot/nfeske_test/bin/x86_64/libc/2023-06-15
var/run/depot_remove/genode/depot/nfeske_test/bin/x86_64/bash-minimal/2023-06-28
var/run/depot_remove/genode/depot/nfeske_test/bin/x86_64/init/2023-06-15
var/run/depot_remove/genode/depot/nfeske_test/bin/x86_64/backdrop/2023-06-15
var/run/depot_remove/genode/depot/nfeske_test/bin/x86_64/chroot/2023-06-15
var/run/depot_remove/genode/depot/nfeske_test/bin/x86_64/terminal/2023-06-15
var/run/depot_remove/genode/depot/nfeske_test/bin/x86_64/fs_report/2023-06-15
[... cut ...]
var/run/depot_remove/genode/depot/nfeske_test/pkg/chroot/2023-06-15
var/run/depot_remove/genode/depot/nfeske_test/pkg/motif_decorator/2023-06-15
var/run/depot_remove/genode/depot/nfeske_test/pkg/system_shell/2023-06-28
```

I admittedly haven't looked into test in detail. By glancing at the log output, I seems that the test 3 is supposed to remove the pkg "fonts_fs" by the `[depot_user]`, which is what seems to happen. The removal of the dependency `raw/fonts_fs` seems plausible to me. But the test apparently expects that this archive remains untouched.

chelmuth, 2023-08-24:
Merged in Genode 23.08.

nfeske, 2023-08-25:
> concerning test 3 failure, please try fixup [4adc825](https://github.com/a-dmg/genode/commit/4adc8252ede8f707bf7b86275f495d2f9501248d) commit and let me know if that works for you?

Yes, the fixup solves the problem indeed. Thanks a lot! I have merged the commits to staging now.

nfeske, 2023-10-25:
In the meantime, the follow-up improvements reached the master branch.

closed 2023-10-25 by nfeske