genodelabs/genode #4869: base: build errors with '-std=gnu++20'

cproc, 2023-05-12:
When building, for example, `init` with `-std=gnu++20`, the following error occurs:

```
In file included from /.../repos/base/include/base/trace/logger.h:17,
                 from /.../repos/base/include/base/thread.h:20,
                 from /.../repos/base/include/base/rpc_server.h:18,
                 from /.../repos/base/include/base/child.h:17,
                 from /.../repos/os/src/lib/sandbox/child.h:19,
                 from /.../repos/os/src/lib/sandbox/child.cc:18:
/.../repos/base/include/base/trace/buffer.h: In member function 'void Genode::Trace::Simple_buffer::_commit(
Genode::size_t, WRAP_FUNC&&)':
/.../repos/base/include/base/trace/buffer.h:188:25: error: '++' expression of 'volatile'-qualified type is deprecated [-Werror=volatile]
  188 |                         _num_entries++;
      |                         ^~~~~~~~~~~~
```

and errors related to `-Werror=deprecated-enum-enum-conversion`, for example:

```
/.../repos/base/include/base/child.h: In static member function 'static Genode::Ram_quota Genode::Child::env_ram_quota()':
/.../repos/base/include/base/child.h:722:60: error: arithmetic between different enumeration types 'Genode::Cpu_session::<unnamed enum>' and 'Genode::Pd_session::<unnamed enum>' is deprecated [-Werror=deprecated-enum-enum-conversion]
  722 |                         return { Cpu_connection::RAM_QUOTA + Pd_connection::RAM_QUOTA +
      |                                  ~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~
```
```
In file included from /.../repos/base/include/base/capability.h:20,
                 from /.../repos/base/include/base/signal.h:23,
                 from /.../repos/base/include/vm_session/vm_session.h:21,
                 from /.../repos/os/src/lib/sandbox/child.cc:15:
/.../repos/base/include/base/rpc.h: In instantiation of 'struct Genode::Rpc_function_msg_size<Genode::Datasp
ace::Rpc_size, Genode::RPC_REPLY>':
/.../repos/base/include/base/rpc_client.h:131:68:   required from 'typename IF::Ret_type Genode::Capability< <template-parameter-1-1> >::_call(typename IF::Client_args&) const [with IF = Genode::Dataspace::Rpc_size; RPC_INTERFACE = Genode::Dataspace; typename IF::Ret_type = long unsigned int; typename IF::Client_args = Genode::Meta::Empty]'
/.../repos/base/include/base/capability.h:166:20:   required from 'typename IF::Ret_type Genode::Capability< <template-parameter-1-1> >::call() const [with IF = Genode::Dataspace::Rpc_size; RPC_INTERFACE = Genode::Dataspace; typename IF::Ret_type = long unsigned int]'
/.../repos/base/include/base/rpc_client.h:202:33:   required from 'typename IF::Ret_type Genode::Rpc_client< <template-parameter-1-1> >::call(ARGS&& ...) [with IF = Genode::Dataspace::Rpc_size; ARGS = {}; RPC_INTERFACE = Genode::Dataspace; typename IF::Ret_type = long unsigned int]'
/.../repos/base/include/dataspace/client.h:28:53:   required from here
/.../repos/base/include/base/rpc.h:277:30: error: arithmetic between different enumeration types 'Genode::Rpc_msg_payload_size<Genode::Dataspace::Rpc_size, false, true>::<unnamed enum>' and 'Genode::Rpc_retval_size<long unsigned int>::<unnamed enum>' is deprecated [-Werror=deprecated-enum-enum-conversion]
  276 |                 enum { Value = Rpc_msg_payload_size<RPC_FUNCTION, false, true>::Value
      |                                                                                 ~~~~~
  277 |                              + Rpc_retval_size<typename RPC_FUNCTION::Ret_type>::Value
      |                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```
```
In file included from /.../repos/base/include/dataspace/client.h:18,
                 from /.../repos/base/include/pd_session/client.h:18,
                 from /.../repos/base/include/base/service.h:18,
                 from /.../repos/base/include/base/child.h:19:
/.../repos/base/include/base/rpc_client.h: In instantiation of 'struct Genode::Rpc_function_caps_out<Genode::Dataspace::Rpc_size>':
/.../repos/base/include/base/rpc_client.h:132:57:   required from 'typename IF::Ret_type Genode::Capability< <template-parameter-1-1> >::_call(typename IF::Client_args&) const [with IF = Genode::Dataspace::Rpc_size; RPC_INTERFACE = Genode::Dataspace; typename IF::Ret_type = long unsigned int; typename IF::Client_args = Genode::Meta::Empty]'
/.../repos/base/include/base/capability.h:166:20:   required from 'typename IF::Ret_type Genode::Capability< <template-parameter-1-1> >::call() const [with IF = Genode::Dataspace::Rpc_size; RPC_INTERFACE = Genode::Dataspace; typename IF::Ret_type = long unsigned int]'
/.../repos/base/include/base/rpc_client.h:202:33:   required from 'typename IF::Ret_type Genode::Rpc_client< <template-parameter-1-1> >::call(ARGS&& ...) [with IF = Genode::Dataspace::Rpc_size; ARGS = {}; RPC_INTERFACE = Genode::Dataspace; typename IF::Ret_type = long unsigned int]'
/.../repos/base/include/dataspace/client.h:28:53:   required from here
/.../repos/base/include/base/rpc_client.h:60:88: error: arithmetic between different enumeration types 'Genode::Rpc_caps_out<Genode::Meta::Empty>::<unnamed enum>' and 'Genode::Cap_return<long unsigned int>::<unnamed enum>' is deprecated [-Werror=deprecated-enum-enum-conversion]
   60 |                 enum { Value = Rpc_caps_out<typename RPC_FUNCTION::Server_args>::Value +
      |                                                                                  ~~~~~~^
   61 |                                Cap_return  <typename RPC_FUNCTION::Ret_type>::Value}; };
      |                                ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~     
```

chelmuth, 2023-05-15:
> ```
> In file included from /.../repos/base/include/base/trace/logger.h:17,
>                  from /.../repos/base/include/base/thread.h:20,
>                  from [...]

`_num_entries = _num_entries + 1` is the proposed fix for this.

> ```
> /.../repos/base/include/base/child.h: In static member function 'static Genode::Ram_quota Genode::Child::env_ram_quota()':
> /.../repos/base/include/base/child.h:722:60: error: arithmetic between [...]
[...]

We could replace all unnamed enums by `static constexpr int`, which complies in my tests.

nfeske, 2023-05-30:
Fixed in master.

closed 2023-05-30 by nfeske