genodelabs/genode #4870: icu: build error with `-std=gnu++20`

cproc, 2023-05-12:
When building `lib/icu` with `-std=gnu++20` as default, the following error occurs:

```
/.../contrib/icu-62f13b0976e6f74904ad9dfc82f1bc686fb36b6e/src/lib/icu/source/i18n/basictz.cpp: In member function 'void icu_51::BasicTimeZone::getTimeZoneRulesAfter(UDate, icu_51::InitialTimeZoneRule*&, icu_51::UVector*&, UErrorCode&) const':
/.../contrib/icu-62f13b0976e6f74904ad9dfc82f1bc686fb36b6e/src/lib/icu/source/i18n/basictz.cpp:409:37: error: return type of 'virtual UBool icu_51::TimeArrayTimeZoneRule::operator==(const icu_51::TimeZoneRule&) const' is not 'bool'
  409 |                 if (*(tzt0.getTo()) == *tar) {
      |                     ~~~~~~~~~~~~~~~~^~~~~~~
/.../contrib/icu-62f13b0976e6f74904ad9dfc82f1bc686fb36b6e/src/lib/icu/source/i18n/basictz.cpp:409:37: note: used as rewritten candidate for comparison of 'const icu_51::TimeZoneRule' and 'const icu_51::TimeArrayTimeZoneRule'
```

ssumpf, 2023-05-15:
@cproc: I have updated `icu` to 73.1 on my `gcc-12` branch I cannot find anything that uses it (`LIBS` or `used_api`). How could I test this? In case we need to 51 version for some reason, I can fix this easily also.

chelmuth, 2023-05-16:
:+1: for the removal

ssumpf, 2023-05-16:
> +1 for the removal

Will do.

ssumpf, 2023-05-16:
@chelmuth: bfe3f1b removes icu.

commit dffdc581 (2023-05-16): libport: remove icu library

This was required by QtWebKit in the past, the QtWebEngine ships its own
version. No other code depends on this. Revive if needed.

fixes #4870
closed 2023-05-30 by chelmuth