genodelabs/genode #4874: sculpt_manager: build error with `-std=gnu++20`

cproc, 2023-05-12:
When building `app/sculpt_manager` with `-std=gnu++20` as default, the following error occurs:

```
In file included from /.../repos/gems/src/app/sculpt_manager/model/runtime_state.h:25,
                 from /.../repos/gems/src/app/sculpt_manager/graph.h:33,
                 from /.../repos/gems/src/app/sculpt_manager/graph.cc:14:
/.../repos/gems/src/app/sculpt_manager/model/component.h: In lambda function:
/.../repos/gems/src/app/sculpt_manager/model/component.h:157:90: error: expected ',' or '...' before 'requires'
  157 |                                 runtime.with_optional_sub_node("requires", [&] (Xml_node requires) {
      |                                                                                          ^~~~~~~~
/.../repos/gems/src/app/sculpt_manager/model/component.h: In lambda function:
/.../repos/gems/src/app/sculpt_manager/model/component.h:158:73: error: expected '{' before ')' token
  158 |                                         _update_routes_from_xml(requires); });
      |                                                                         ^
```

nfeske, 2023-05-15:
Apparently `requires` is now a keyword. Renaming the argument name to `req` (https://github.com/genodelabs/genode/commit/f53d15fc131c3752a8daf55910f9c3ce19a7c986) should hopefully fix it.

cproc, 2023-05-15:
Yes, that fixes it.

closed 2023-05-30 by chelmuth