genodelabs/genode #4879: arm_v8a: strange floating-point errors with GCC-12

chelmuth, 2023-05-15:
Since GCC-12, `make run/ieee754` fails on on arm_v8a for *hw* and *foc* alike.

arm_v8a.rpi3.foc.ieee754.log
```
[2023-05-14 07:27:54] Line 13 of output is unexpected
[2023-05-14 07:27:54]  expected: 'FLT_EPSILON = 1.19209e-07 = 0x1p-23'
[2023-05-14 07:27:54]  got:      'FLT_EPSILON = nan = nan'
[2023-05-14 07:27:54] 
[2023-05-14 07:27:54] Line 15 of output is unexpected
[2023-05-14 07:27:54]  expected: 'LDBL_EPSILON = 1.92593e-34 = 0x1p-112'
[2023-05-14 07:27:54]  got:      'LDBL_EPSILON = 0 = 0x1.fffffffffffffffep-16431'
```

arm_v8a.rpi3.hw.ieee754.log
```
[2023-05-14 07:28:00] Line 13 of output is unexpected
[2023-05-14 07:28:00]  expected: 'FLT_EPSILON = 1.19209e-07 = 0x1p-23'
[2023-05-14 07:28:00]  got:      'FLT_EPSILON = nan = nan'
[2023-05-14 07:28:00] 
[2023-05-14 07:28:00] Line 15 of output is unexpected
[2023-05-14 07:28:00]  expected: 'LDBL_EPSILON = 1.92593e-34 = 0x1p-112'
[2023-05-14 07:28:00]  got:      'LDBL_EPSILON = 0 = 0x1.fffffffffffffffep-16431'
```

arm_v8a.imx8q_evk.hw.ieee754.log
```
[2023-05-14 08:50:51] Line 13 of output is unexpected
[2023-05-14 08:50:51]  expected: 'FLT_EPSILON = 1.19209e-07 = 0x1p-23'
[2023-05-14 08:50:51]  got:      'FLT_EPSILON = nan = nan'
[2023-05-14 08:50:51] 
[2023-05-14 08:50:51] Line 15 of output is unexpected
[2023-05-14 08:50:51]  expected: 'LDBL_EPSILON = 1.92593e-34 = 0x1p-112'
[2023-05-14 08:50:51]  got:      'LDBL_EPSILON = 0 = 0x1.fffffffffffffffep-16431'
```

arm_v8a.virt_qemu_arm_v8a.hw.ieee754.log
```
[init -> test4] Variable expression 1 + DBL_MIN = 1.0000000000000002221
Kernel: Unknown cpu exception EC=60 ISS=1000 ip=0xffffffc00000d988
```

Also, `make -C build/arm_v8a/ run/depot_autopilot TEST_PKGS=test-xml_generator` breaks at https://github.com/genodelabs/genode/blob/2e76374a357c4d54ee932606f6267f91e006e102/repos/base/src/test/xml_generator/main.cc#L237-L242 The generated floating-point string is `" 4 == 2 + 2 == 0.0 "` but should reflect `4.0`.

cproc, 2023-05-15:
Maybe another FPU state problem like #3723?

cproc, 2023-05-15:
I'm starting git bisect. Maybe the GCC commit that triggered the problem can give a hint.

ssumpf, 2023-05-15:
@cproc: FYI: The problem can be reproduced with a simple:
```
Genode::log("value: ", 4.0f);
```
which will produce `value: 0.0`

cproc, 2023-05-15:
git bisect just finished and found the auto-vectorization commit ([1]).

Building the test alone with `-fno-tree-vectorize` did not make it work, but building everything with the option did.

[1] https://github.com/gcc-mirror/gcc/commit/2b8453c401b699ed93c085d0413ab4b5030bcdb8

cproc, 2023-05-15:
Maybe `setjmp/longjmp` and/or `_jmp_slot` need to save and restore FPU state, too? x86_64 saves and restores some of it.

ssumpf, 2023-05-15:
> `-fno-tree-vectorize`

Does it for the `Genode::log("value: 4.0f")` as well, and therefore, also for `make -C build/arm_v8a/ run/depot_autopilot TEST_PKGS=test-xml_generator`. I will check the `_jmp_slot` next. Good job! 
`

ssumpf, 2023-05-15:
@cproc: 0c54281 fixes the issue for me without `-fno-tree-vectorize`.

ssumpf, 2023-05-15:
@cproc: `setjmp/longjmp` are fine, they save/restore the registers.

cproc, 2023-05-16:
> @cproc: [0c54281](https://github.com/genodelabs/genode/commit/0c5428152ef536991604d3c67822d1968a51ac26) fixes the issue for me without `-fno-tree-vectorize`.

Yes, for me as well (tested with virt_qemu_arm_v8a).

ssumpf, 2023-05-16:
@chelmuth: I think we can mark this as fixed. Thanks to @cproc this went a lot quicker than anticipated.

chelmuth, 2023-05-16:
Great! Will merge the commit.

commit 5e0af427 (2023-05-16): ldso: save floating point arguments on arm_v8a (GCC 12)

Apparently we need to save the floating point arguments when entering
the linker in case "-fno-tree-vectorize" is at work. Otherwise this
registers get corrupted during jump slot relocation.

Fixes #4879
closed 2023-05-30 by chelmuth