genodelabs/genode #4885: pc_uart_drv: build error with `-std=gnu++20`

cproc, 2023-05-16:
When building `drivers/uart` with `-std=gnu++20` as default, the following error occurs:

```
In file included from /.../repos/os/src/drivers/uart/uart_component.h:29,
                 from /.../repos/os/src/drivers/uart/main.cc:20:
/.../repos/os/src/drivers/uart/spec/x86/uart_driver.h: In member function 'void Uart::Driver::_init_comport(Genode::size_t)':
/.../repos/os/src/drivers/uart/spec/x86/uart_driver.h:90:57: error: '--' expression of 'volatile'-qualified type is deprecated [-Werror=volatile]
   90 |                         for (volatile int i = 10000000; i--; );
      |
```

cproc, 2023-05-16:
This way does not work either. How to fix it?

```
In file included from /.../repos/os/src/drivers/uart/uart_component.h:29,
                 from /.../repos/os/src/drivers/uart/main.cc:20:
/.../repos/os/src/drivers/uart/spec/x86/uart_driver.h: In member function 'void Uart::Driver::_init_comport(Genode::size_t)':
/.../repos/os/src/drivers/uart/spec/x86/uart_driver.h:90:66: error: using value of assignment with 'volatile'-qualified left operand is deprecated [-Werror=volatile]
   90 |                         for (volatile int i = 10000000; i > 0; i = i - 1);
      |                                                                ~~^~~~~~~
```

chelmuth, 2023-05-16:
The following patch compiles, but is it correct to use this rule-of-thumb loop as *delay function*?

```diff
+++ b/repos/os/src/drivers/uart/spec/x86/uart_driver.h
@@ -87,7 +87,7 @@ class Uart::Driver : public Uart::Driver_base
                void _init_comport(size_t baud)
                {
                        _outb<LCR>(0x80u);  /* select bank 1 */
-                       for (volatile int i = 10000000; i--; );
+                       for (volatile int i = 10000000; i; ) i = i - 1;
```

closed 2023-05-30 by chelmuth