genodelabs/genode #4913: sel4: convert io memory on demand

alex-ab, 2023-05-31:
During boot-up core converts all untyped I/O memory eagerly. On x86 and qemu this takes quite a long time for memory never used. On demand conversion should avoid the overhead.

chelmuth, 2023-05-31:
I merged the commits to staging. Could the issue be marked fixed?

alex-ab, 2023-05-31:
> I merged the commits to staging. Could the issue be marked fixed?

Yes, marked it.

nfeske, 2023-05-31:
The change in https://github.com/genodelabs/genode/commit/24cd9cfa5e1ec73a288796bb8fafd6d5795b168c looks good to me. I was slightly puzzled that the interface change of `Io_mem_session_component::_map_local` does not affect other base platforms. But then, I saw the fixup https://github.com/genodelabs/genode/commit/7986f9fafe44c199df2a24d258932ccace320043. All fine by me.

commit af10982e (2023-05-31): sel4: convert io memory on demand

instead all during boot. On x86 and qemu this takes quite a while until
the system is booted.

Fixes #4913
commit b349dd9c (2023-06-16): sel4: convert io memory on demand

instead all during boot. On x86 and qemu this takes quite a while until
the system is booted.

Fixes #4913
closed 2023-06-16 by nfeske