genodelabs/genode #4915: base: support inhibiting of tracing via `Genode::Thread`

cproc, 2023-06-01:
While experimenting with a modified `trace_logger` test on base-linux I noticed that the solution from issue #4607 has the side-effect that trace events of unrelated threads can get lost while tracing is temporarily inhibited via the global `Genode::inhibit_tracing` variable. The problem can be solved by supporting the inhibiting of tracing on a per-thread basis via a `Thread::inhibit_tracing()` method that can be used with `Thread::myself()`.

cproc, 2023-06-01:
Fixed by b891f57

nfeske, 2023-06-01:
This change seems to break components using the lx_hybrid library, e.g., fb_sdl. I vaguely guess that the lx_hybrid library may leave the `Thread::_trace_control` uninitialized whereas the new `inhibit_tracing` and `tracing_inhibited` methods expect a valid pointer.

cproc, 2023-06-01:
> This change seems to break components using the lx_hybrid library, e.g., fb_sdl. I vaguely guess that the lx_hybrid library may leave the `Thread::_trace_control` uninitialized whereas the new `inhibit_tracing` and `tracing_inhibited` methods expect a valid pointer.

Yes, and `Thread::myself()` also appears to return `nullptr` for the main thread whereas other platforms create a `Main_thread` object. If tracing of lx_hybrid components is not supported anyway, a simple fix would be to set the global `inhibit_tracing` variable to `true` again after it has been set to `false` in `Constructor_component::construct()`. Fixup commit f8c570a uses the lx_hybrid-specific `Genode::call_global_static_constructors()` implementation for that.

nfeske, 2023-06-02:
Upon closer inspection, I sense that we went too far into a wrong direction.

Let's recap that the fix of issue #4607 solved the observed problem that the passing of a tracepoint may sometimes trigger side effects that, in turn, execute code that involves other tracepoints. So the root of the problem is the nested passing of tracepoints. The fix of issue #4607 did not address this general problem but is a symptomatic fix that solved the observed special case.

Now, @cproc discovered that this symptomatic fix introduced a new problem. Unfortunately, the proposed solution for this new problem introduces a lot of complexity in the passing. In particular, it extends the public `Thread` API with methods that are not meaningful for the API user (`inhibit_tracing`, `tracing_inhibited`). More problematic is added public accessor that returns a raw pointer (`trace_control_from_base`), which is the opposite direction of where I want to go. Fixup https://github.com/genodelabs/genode/commit/f8c570a50008309a6a3304214db676e0838a64bf does not make the situation any better. It raises eyebrows because tracing is unrelated to global constructors.

Please let us consider solving the root problem of issue #4607 instead. Maybe I'm oversimplifying things but how about detecting the nested attempt to call `Trace::Logger::_evaluate_control` in the first place? I think of introducing a `bool masked` to the `Trace::Logger` that is set while `_evaluate_control` calls functions with side effects (detach/attach), and is evaluated in the condition at `Logger::log` (https://github.com/genodelabs/genode/blob/master/repos/base/include/base/trace/logger.h#L84) or at the beginning of `_evaluate_control`. With this approach, we can revert 3f1dff4 and the `Thread` API remains unchanged.

cproc, 2023-06-02:
The original observed problem was that `Region_map_mmap::attach()/detach()` was called recursively because it passes tracepoints and `Trace::Logger::_evaluate_control()` attaches/detaches dataspaces in some conditions. Maybe I can try to detect the recursion in `Region_map_mmap()` and let the affected function fail in that case.

nfeske, 2023-06-02:
Ideally, the `Region_map_mmap` would avoid taking the mutex in the nested case and let the operation succeed.

chelmuth, 2023-06-02:
Just *opening up* the mutex involves the risk of further surprises as the mutex-holder just preempts itself a the place of the trace point. In my opinion, it may be better to analyze the `attach` and `detach` methods how the critical sections could be minimized to not longer include paths with trace points. Unfortunately, this can only comprise the current implementation of used functions, which may change in the future and also introduce trace points.

A start could be to move calls to `_dataspace_size()` and `_dataspace_writeable` before the critical section.

cproc, 2023-06-02:
I moved the calls to `_dataspace_size()`, `_dataspace_writeable()` and `_dataspace_fd()` before the critical section in WIP commit f65fbc5 in two of the three cases, but I'm not sure how to do it in the case with the region loop (marked with XXX).

cproc, 2023-06-02:
If the `fd` and `writeable` dataspace attributes stay constant over time, maybe they could be cached in the `Region` object and retrieved from there in the region loop case?

chelmuth, 2023-08-24:
I removed the fixed label for the moment. @cproc please update the state of this issue.

commit 9df22554 (2023-06-01): base: support inhibiting of tracing via `Genode::Thread`

Fixes #4915