genodelabs/genode #4917: Debug monitor based on the sandbox API

nfeske, 2023-06-07:
The existing GDB monitor has a number of limitations that we need to overcome to make on-target debugging a regular feature of Genode and Sculpt OS. The most visible problems are:

* The GDB monitor supports only one component as debugging target, which makes the debugging of scenarios where components closely interact difficult. (illustrated by issue #4138)
* The existing implementation re-uses the gdbserver code and thereby inherits many POSIX peculiarities that must be stubbed for Genode, yet make the overall implementation complex. (illustrated by issue #4143)

On our [road map](https://genode.org/about/road-map#August_-_Release_23.08), we have scheduled the revision of Genode's GDB support for the August release. Two key ideas motivate a new approach:

First, by using Genode's sandbox API as foundation for a new debug monitor, we become able to use the monitor as drop-in replacement for init, potentially going as far as using the monitor for Sculpt's runtime subsystem. And second, GDB supports the debugging of multiple processes (called inferiors) within one session, which would in principle allow us to inspect and debug component compositions.

In real-world scenarios like Sculpt's runtime, we deliberately want/need to restrict the debugging to a few selected components. This calls for the support of a mix of monitored and regular components hosted side by side. Given this requirement, the sandbox API must be enhanced to support the selective interception of PD and CPU sessions.

chelmuth, 2023-06-09:
I noted that the monitor component is not yet available as depot archive. Would you mind to amend the current implementation by a src recipe?

nfeske, 2023-06-09:
> Would you mind to amend the current implementation by a src recipe?

Added by fixup https://github.com/genodelabs/genode/commit/8460122c9441b4a6f6240d068f6c930cc5fb1394

nfeske, 2023-06-16:
The initial version of the monitor component has entered the master branch.

For the steps towards the complete feature parity with the old gdb_monitor, let us open dedicated issues.

@cproc, would you like to pick up the torch from this point on?

I think the next sensible steps could be:

* Management of thread-pause states (adhering to the `<policy>` `stop` and `wait` attributes). Adjusting the README once that works.
* Reflecting the thread state for NOVA. Currently, the register content is only available for base-hw because the `Cpu_thread::state` function returns values on non-paused threads on this kernel. On NOVA, I presume that we need to pause the thread first.
* Complementing the debugging features (single step, break points) as tested by the existing gdb_monitor.run script, but incorporating the use of the `inferior` selection. For filling the gaps of the GDB-protocol implementation, it is useful to first gather a trace of the old gdb_monitor's interaction (e.g., by instrumenting the UART driver).
* Responding to CPU exceptions (e.g., division by zero).
* Investigating the use of GDB's nonstop mode, which would allow us to selectively start and stop individual inferiors (or even individual threads?).
* Adding support for arm_v8a.
* Removing the original GDB monitor from the ports repository.
* Potentially supporting kernels other than NOVA and base-hw, e.g., seL4.

closed 2023-06-16 by nfeske