genodelabs/genode #4918: genode_c_api: MAC address reporter utility

chelmuth, 2023-06-09:
NIC drivers should optionally report their discovered and programmed NIC devices along with the effective MAC addresses. To consolidate this feature I propose the following addition to the Genode C API.

```c++
/*
 * \brief  C-API Genode MAC address reporter utility
 * \author Christian Helmuth
 * \date   2023-06-06
 */

/*
 * Copyright (C) 2023 Genode Labs GmbH
 *
 * This file is part of the Genode OS framework, which is distributed
 * under the terms of the GNU Affero General Public License version 3.
 */

#ifndef _GENODE_C_API__MAC_ADDRESS_REPORTER_H_
#define _GENODE_C_API__MAC_ADDRESS_REPORTER_H_

#include <genode_c_api/base.h>

#ifdef __cplusplus

#include <util/xml_node.h>

void genode_mac_address_reporter_init(Genode::Env &, Genode::Allocator &);

void genode_mac_address_reporter_config(Genode::Xml_node const &);

#endif

#ifdef __cplusplus
extern "C" {
#endif

void genode_mac_address_reporter_report(char const * name,
                                        unsigned char const * addr);

#ifdef __cplusplus
}
#endif

#endif /* _GENODE_C_API__MAC_ADDRESS_REPORTER_H_ */
```

I'd like to hear your opinion in particular to the following aspects.
- Is it okay to keep the initialization function in the C++-only interface?
- Is the `unsigned char const *` type to lax for reported MAC addresses?

nfeske, 2023-06-09:
> Is it okay to keep the initialization function in the C++-only interface?

Yes.

> Is the `unsigned char const *` type to lax for reported MAC addresses?

It doesn't set a good precedent. Defining a struct with an array of 6(?) bytes would leave no room for wrong interpretation.

To me, the name `genode_mac_address_reporter_report` looks a bit forced. Would you consider `genode_register_mac_address` as an alternative?

Out of curiosity, can a MAC possibly disappear?

chelmuth, 2023-06-12:
> > Is the `unsigned char const *` type to lax for reported MAC addresses?
> 
> It doesn't set a good precedent. Defining a struct with an array of 6(?) bytes would leave no room for wrong interpretation.

I was half afraid of exactly this answer ;-) and will change the interface accordingly.

> To me, the name `genode_mac_address_reporter_report` looks a bit forced. Would you consider `genode_register_mac_address` as an alternative?

Good point, I'd prefer `genode_mac_address_register` then.

> 
> Out of curiosity, can a MAC possibly disappear?

I could imagine dynamic MAC addresses in the following cases.
- virtual-function PCI devices
- the address is changed by the system (aka set_mac_address)
- drivers for some kind of dynamic devices that become available (or disappear) after some kind of hot-plugging event (Thunderbolt?) but I have no real example

As we do not support the above cases currently, I did not add `unregister`.

chelmuth, 2023-06-13:
Commit a9c9b99bbd9294e94c22a8492a0f1b904dc311ac implements the utility. Now, we should adapt all drivers to this feature before we close this issue.

nfeske, 2023-06-16:
Merged to master.

chelmuth, 2023-06-16:
Let's keep open until all drivers are adapted to the utility.

nfeske, 2023-06-16:
Related to #4133.

nfeske, 2023-07-14:
Fixed in master.

closed 2023-07-14 by nfeske