genodelabs/genode #4922: depot_autopilot: simplify definition of success criterions

nfeske, 2023-06-13:
As the rough plan to support automated testing in Goa is shaping up, it makes sense to share one convention about expressing the success criterion for a package under test between the depot_autopilot and Goa.

For reference, the current scheme looks like [this](https://github.com/genodelabs/genode/blob/master/repos/os/recipes/pkg/test-rom_filter/runtime#L5-L19) example in practice. I suggest the following simplifications:

* Drop the intermediate `<events>` node.
* Replace `<log meaning="succeeded">` by `<success>`.
* Replace `<log meaning="failed">` by `<fail>`.
* Replace `<timeout meaning="failed" sec="20" />` by an `after_seconds` attribute of the `<success>` or `<fail>` nodes.

To illustrate the change, the following example (based on the one linked above)
```
<runtime ...>
  ...
  <events>
    <timeout meaning="failed" sec="20" />
    <log meaning="succeeded">
      [init -> rom_logger] ROM 'generated':
      ...
      [init -> dynamic_rom] xray: change (finished)
    </log>
    <log meaning="failed">Error: </log>
  </events>
  ...
</runtime>
```

... would become ...

```
<runtime ...>
  ...
  <fail after_seconds="20'/>
  <success>
    [init -> rom_logger] ROM 'generated':
    ...
    [init -> dynamic_rom] xray: change (finished)
  </success>
  <fail>Error: </fail>
  ...
</runtime>
```

If we agree to this simplification, the implementation would take three steps:

1. Adding the new scheme in addition to the original one to the depot_autopilot
2. Gradually simplifying our `pkg/test-<name>/runtime` files. While doing this, reconsider the use of `init` for all tests that host only a single component. In these cases, the test program can be started directly as `<runtime binary="..."`>.
3. Once the new scheme is consistently applied, remove the old scheme from depot autopilot.

At this point, the way would be cleared to re-use the same convention in Goa.

nfeske, 2023-06-13:
Thanks @jschlatow for the immediate feedback! Happy to see your overall approval.

> *  My symmetry-seeking mind suggests <success> be rather named <succeed>.
> * I'd like to moot an optional log_prefix attribute.

I'm with you on both points.

@m-stein as you came up with the existing scheme, would you agree to the new one as well?

m-stein, 2023-06-13:
@nfeske I'm fine with that. Thanks for the good presentation and for the involvement! Merely as a suggestion in case you didn't already thought of it: Considering the number of tests, the transition of the test files _might_ be easier or even be reduced to one commit using tooling like xpath or xsltproc.

nfeske, 2023-06-14:
Thank you Martin for the quick and welcoming response.

> [...] the transition of the test files might be easier or even be reduced to one commit using tooling like xpath or xsltproc

This could indeed save many hours of manual text-wrangling work. I like the idea but XSLT scares me.

You may have seen this coming: Would you like to try coming up with an xslt script? ;-)

m-stein, 2023-06-14:
To my shame, I ran into that trap without any suspicion ^_^ I put it on my todo list and will give it a try.

m-stein, 2023-06-24:
@jschlatow I don't know what the meaning of ``log_prefix`` is. Could you give me a brief explanation?

m-stein, 2023-06-24:
Except for the ``log_prefix``, I've implemented the described approach in  af7f90f661. Any opinions?

m-stein, 2023-06-24:
PS: I have also already adapted the runtime files in the other repos and would push those commits after the review stage.

m-stein, 2023-06-25:
I just noticed that there is a package in allwinner that is named test-* although not being a valid depot-autopilot test (recipes/pkg/test-camera-pinephone). Should we consider renaming it?

World commit:
https://github.com/genodelabs/genode-world/commit/7f4ea7cc457b1dacf5ccd7dc61beffb804e949c5

jschlatow, 2023-06-26:
> @jschlatow I don't know what the meaning of `log_prefix` is. Could you give me a brief explanation?

The idea was to use this as a poor-mans-grep: When provided, only the lines from the log output that start with the log_prefix-value will be used when comparing against the expected output.

m-stein, 2023-06-26:
@jschlatow Thanks for the explanation! If there are no objections, I would implement the log_prefix in a separate commit as it requires adapting the log processing, a somewhat complicated part of the depot_autopilot. That said, the above mentioned commits would be ready for merging from my side.

m-stein, 2023-06-27:
@jschlatow I'm almost done with the log prefix but there are two details about this feature that I would not like to decide by myself:
* Should the prefix also affect what is forwarded to the Depot Autopilots log? If yes, only those of the test-log lines that start with the prefix would enter the nightly log files of the Depot Autopilot.
* Should pattern matching simply assume that each pattern line starts with the prefix? If yes, the prefix must be removed from the pattern when adding ``log_prefix``, but removing it would not require adding wildcards. If no, the prefix might be present in the pattern but in order to remove it, one might have to add a wildcards (multi-line patterns).

m-stein, 2023-06-27:
@jschlatow Commit 3ae3b69ce2 implements what I considered the most suitable approach (a "no" to both of my questions). It is ready for merging, but I can also implement another flavor if desired. If you agree with that commit I would appreciate you creating some regular "stress" for the ``log_prefix`` feature ;-)

jschlatow, 2023-06-27:
@m-stein I agree with the double-"no".

I just tried stressing the `log_prefix` feature by modifying `test-fs_tool`:
- I was able to only match the lines logged by the report_rom component. Very nice!
- I was also able to only compare those lines that start with `[init -> test -> remove] Warning`.
- However, I couldn't manage to extract those lines starting with `[init -> report_rom]   <file`. I believe the log_prefix-string is not checked for any escape sequences? Cause the `<` must be escaped as `&lt;`.

nfeske, 2023-06-28:
> I just noticed that there is a package in allwinner that is named test-* although not being a valid depot-autopilot test (recipes/pkg/test-camera-pinephone). Should we consider renaming it?

Suitability for the use with depot-autopilot is not a prerequisite for prefixing a pkg name with `test`. All depot-autopilot tests are prefixed with `test` but not all test pkgs need to be depot-autopilot tests.

m-stein, 2023-06-28:
@nfeske Ok, thanks for the clarification! It's fine with me to keep it.

m-stein, 2023-06-28:
@jschlatow Thanks for your fast hands on! You're right, I did not consider escape sequences. I'll provide you with a fix soon.

m-stein, 2023-06-29:
@jschlatow Could you give this updated commit series a try?
449cb312b9 depot_autiopilot: consider log_prefix attribute
6776a8eab9 depot_autopilot: simplify success-criterion syntax
4168aa1b92 xml_unquoted: fix bug with string of length 0

The bugfix for Xml_unquoted is necessary.

jschlatow, 2023-06-29:
@m-stein I believe there is still a corner case. It seems to work if the filtered lines are consecutive in the original log output. However, the following modification of test-fs_tool fails:

```
	<fail after_seconds="20"/>
	<succeed log_prefix=""[init -> report_rom]   &lt;dir>
			[init -> report_rom]   &lt;dir path="/fs/items">
			[init -> report_rom]   &lt;dir path="/fs/new/file/at/subdir">
	</succeed>
```

m-stein, 2023-06-29:
@jschlatow I think your example doesn't fit, because even after applying the log_prefix, the two pattern lines are not consecutive. However, you're right that something like this doesn't work:

```
<succeed log_prefix="[init -> test-log]">
	[init -> test-log] Very long messages:
	[init -> test-log] 3.....................................................................................................................................................................................................................................4
</succeed>
```
When the output is this:
```
...
[init -> test-log] Very long messages:
[init -> test-log -> log] 1........................................................................................................................................2
[init -> test-log] 3........................................................................................................................................4
...
```
I've update the top-most commit to 040aedf0ee9f64f49d2cb4a71d3d082d7573c005 in order to fix that issue.

m-stein, 2023-06-29:
PS: If you want your test-fs_tool example to work you can add a wildcard (and fix the type with the quotes):
```
<succeed log_prefix="[init -> report_rom]   &lt;dir">
		[init -> report_rom]   &lt;dir path="/fs/items">*
		[init -> report_rom]   &lt;dir path="/fs/new/file/at/subdir">
</succeed>
```

jschlatow, 2023-06-29:
> I think your example doesn't fit, because even after applying the log_prefix, the two pattern lines are not consecutive.

Sorry, but I don't get why. Can you explain?

> I've update the top-most commit to 040aedf in order to fix that issue.

With that commit, my test-fs_tool example works (without any wildcard).

m-stein, 2023-06-29:
@jschlatow Sorry, you're right. I used ``log_prefix="[init -> report_rom]   &lt;"`` instead of ``log_prefix="[init -> report_rom]   &lt;dir"``.

m-stein, 2023-06-30:
@jschlatow While working on the last commits, I found two other issues I wanted to solve. The commits are f6be092dfa and 9ebea7ecc1. What do you think?

jschlatow, 2023-06-30:
> @jschlatow While working on the last commits, I found two other issues I wanted to solve. The commits are [f6be092](https://github.com/genodelabs/genode/commit/f6be092dfa4e6957075f9f11de590fd23e259e21) and [9ebea7e](https://github.com/genodelabs/genode/commit/9ebea7ecc10e07ec62cf388c3f4bf7f1121da8c5). What do you think?

Passing log_prefix and the log output through the same filter functions makes much more sense. I was wondering about the use of Xml_unquoted anyway. I also like that the formatting is preserved when forwarding to the real LOG service. It should be mentioned in the README, though, that color sequences are ignored when matching against the expected output.

nfeske, 2023-07-21:
Thank you @m-stein for implementing this change and the welcome collateral improvements. I merged the commits to staging now.

m-stein, 2023-07-21:
@nfeske Thank you for the merge!

commit 9c766301 (2023-07-21): depot_autopilot: simplify success-criterion syntax

* Removes the <event> tag from all test package runtime files and replaces the
  contained <timeout> and <log> sub-tags with the new tags <succeed> and
  <fail>. If a <succeed> or <fail> tag has a content, it defines a log pattern
  that should be recognized and render the test failed or successful. If a
  <succeed> or <fail> tag has an attribute after_seconds that is not set to 0,
  it defines a timeout after which the test should be rendered failed or
  successful.

* Adapts the Depot Autopilot to support the new syntax in the test-package
 [... cut]
commit c47a6b08 (2023-08-24): depot_autopilot: simplify success-criterion syntax

* Removes the <event> tag from all test package runtime files and replaces the
  contained <timeout> and <log> sub-tags with the new tags <succeed> and
  <fail>. If a <succeed> or <fail> tag has a content, it defines a log pattern
  that should be recognized and render the test failed or successful. If a
  <succeed> or <fail> tag has an attribute after_seconds that is not set to 0,
  it defines a timeout after which the test should be rendered failed or
  successful.

* Adapts the Depot Autopilot to support the new syntax in the test-package
 [... cut]
closed 2023-08-24 by chelmuth