genodelabs/genode #4924: lib/net: enable strict warnings

m-stein, 2023-06-13:


chelmuth, 2023-06-19:
@m-stein I merged commits in this issue plus

b0c673b6a10 tresor: do not use on-disc-layouted structs
65975342cd2 fixup "tresor: do not use on-disc-layouted structs" (fall asleep instead of spinning in assertion)

Please check your merge_to_staging branch as IMO the other commits are obsolete.

chelmuth, 2023-06-27:
> Thanks for the hint! I've re-based my merge_to_staging.
> 
> Is commit [ff89007](https://github.com/genodelabs/genode/commit/ff89007771e7c631bc9436e874344622f705586f) considered valuable? Otherwise, I can remove it.

This change was already merged as e0dc544aa52ff8cc531a506398e7d6959d1c865d in a slightly modified version.

m-stein, 2023-06-28:
@chelmuth Thanks! Good to know.

m-stein, 2023-06-28:
@chelmuth I'm sorry for the inconvenience with the failing tests! Here is the fix-up 568c1df3c0.

commit 9b63e68b (2023-06-19): lib/net: enable strict warnings

Enables strict warnings in the net lib and fixes all related compile errors.

Fixes #4924
commit bf18ffbb (2023-07-14): lib/net: enable strict warnings

Enables strict warnings in the net lib and fixes all related compile errors.

Fixes #4924
closed 2023-07-14 by nfeske