genodelabs/genode #4925: sd_card.run: fix broken config

m-stein, 2023-06-13:


m-stein, 2023-06-13:
I'd opt for either adding 5fb676de46 _and_ adding the test to autopilot or removing the test. Any opinion?

nfeske, 2023-06-14:
Thank you for the repair.

> I'd opt for either adding https://github.com/genodelabs/genode/commit/5fb676de46b72812b62c969951144e00f0b67de1 and adding the test to autopilot or removing the test. Any opinion?

I foresee retiring our original SD-card driver(s) in favor of DDE-Linux-based drivers as already used on i.MX8, allwinner A64, and Zynq. Once we remove the old driver(s), we will retire the run script as well. It is probably not worth adding the run script for automated testing in the intermediate phase.

commit 8ee98fae (2023-06-14): sd_card.run: fix broken config and automate

Fixes #4925
closed 2023-06-16 by nfeske